xoseperez/espurna #1357: Heatbeat

CrispinP, 2018-11-19:
I am using the SONOFFs to tell me if there is power to a circuit or not. I care not about the relay etc.
At the moment I am using a ping to detect when it logs back onto the network and based on that, I know if the lights are on it not.

A better way would be for me to be able to specify a heartbeat topic which publishes epoch time every second. This would allow me to monitor that and, while changing, I know it's working.

Is there something like this already or could it be added?
I'm not seeing something I could use in the current values. uptime does not seem to refresh very often?

thanks.

CrispinP, 2018-11-19:
I see there is a heartbeat but it is not configurable. In general.h it's set at 30 seconds. 
Could this be exposed in the web UI?

xoseperez, 2018-11-19:
This is something specific as you would usually not want to publish this every second. My suggestion is to build your own image modifying the `HEARTBEAT_INTERVAL` setting in the `general.h` file. Then you can decide what values you want to receive by setting or unsetting the `HEARTBEAT_REPORT_*` settings.