xoseperez/espurna #1359: Help Nodemcu serial read write TX RX

munkito, 2018-11-21:
Hello,

I'm trying to read (and, in the future, write) the RX pin of a nodemcu and I can't read anything. I'm sorry but I don't know if there is some missing in my configuration. I can connect to the nodemcu via telnet and I can see all the mqtt messages it is sending (I have a temperature sensor) but I can't see anything in the telnet console. 

In the RX pin I have connected a device wich is sending some data.

Please help.

Thanks!

xoseperez, 2018-12-07:
I'm not sure of what you are trying to do. You said you have a device sending data to the RX of the NodeMCU (like a GPS or a bar scanner or something like that, right?). If that is the case then ESPurna is trying to identify known keywords in that data which it shouldn't since it probably does not know what to do with that data. But you should see errors messages in the telnet session (`ERROR unknown command`).

munkito, 2018-12-09:
Thanks for your answer. To be honest I don't know if what I'm trying to do is possible. The fact is I want to try to use the nodemcu as sniffer, let me explain: I have a Mitsubishi air conditioner, it doesn't have remote control it only has a wall control. I've bought, from Mitsubishi, an addon that permits control it from my smartphone. The problem is that I have more than one machine and if I want to control all of them with my smartphone I have to buy one addon for each one (100€ each one). The addon is connected to the machine through a propietary connector but I've found in internet some people that explain each pin. It has 2 pins used to TX and RX. What I was trying to do is read the data frame of each command to try to implement them on a ESP8266. The telnet console doesn't show any error messages about this, the only error messages I see are about NTP, I think they aren't related with TX RX, doesn't it?

I don't know if you can help me with this. I don't know a lot about electronic, maybe what I'm trying to do is not possible.

Thanks!

xoseperez, 2018-12-09:
You will probably need a custom module to interface the API of the device. Do you have the documentation of the protocol? You can try to use the UART2MQTT module in ESPurna to forward messages from the RX line to your local MQTT broker.

But the proper way to do this will be to first use a development board with a simple serial forwarder from any two pins connected to your air conditioner machine using SoftwareSerial to the Hardware serial port of the board and connected to your computer.

xoseperez, 2018-12-09:
It looks like there is already a library that implements the protocol (https://github.com/SwiCago/HeatPump). It should be straight forward to add support for it.

munkito, 2018-12-09:
Ok, that's good news. How can it be done? Can I do myself? Is there any documentation about how to do it?

Thanks!

mamartins, 2020-02-29:
Was this implemented?

mcspr, 2020-03-01:
> Was this implemented?

Not yet. See https://github.com/SwiCago/HeatPump examples and our specific modules, depending on what exactly needs to happen here e.g. mqtt payload to control things, some local logic, relay bound to a specific action or something else.