xoseperez/espurna #1360: Homey Support

Swiftnesses, 2018-11-21:
Hi,

I heard this was added, can you confirm?

Many thanks!

bakman2, 2018-11-22:
Already supported via mqtt: use the [mqtt broker app](url
https://apps.athom.com/app/nl.scanno.mqttbroker) for homey.

Swiftnesses, 2018-11-22:
Hi,

I don't use MQTT, I wondered if there was a more native implementation available (like Homeyduino).

mcspr, 2018-11-24:
Hello,
Not really sure where you saw it. A cursory glance at Homey app list - there is Tasmota support through MQTT using this https://github.com/robertklep/name.klep.sonoff/, but not ESPurna

If not MQTT, then you can look into utilizing [HTTP api](https://github.com/xoseperez/espurna/wiki/RESTAPI)
Although it is rather limited atm - switching relays directly and for specified time + lights coloring. Something like /device is missing, while the mqtt sends it.

bakman2, 2018-11-24:
@mcspr MQTT = MQTT.