xoseperez/espurna #1361: DS18B20 extended temperature sensors not working

peteeee, 2018-11-22:
Recently switched over from Tasmota to Espurna for my Sonoff WiFi basic switches and multiple DS18B20s.

I am impressed with the Espurna feature set and using less than 500k of memory.

Initial testing with Tasmota firmware had 3 DS18B20 sensors from about 3 meters out to around 10 meters.  This has been working well now for a few months.

Initial testing was connecting 3 DS18B20's on the work bench with short wires.  All three worked fine with ESPURNA 1.13.3.

Extending one of the sensors to around 10 meters took out all of the sensors where I cannot see any of the sensors.

Not sure why the three sensors (extended) worked with Tasmota firmware and no longer work now with ESPurna firmware.

Did a similiar test with one DHT22 sensor. Worked great near the Sonoff and it doesn't work at all extended to around 10 meters.

Valcob, 2018-11-22:
@peteeee could you please share the schematics of how you are connecting the sensors you ll need some pull ups as the cable is quite long also you can try using CAT5 Ethernet cable

AlbertWeterings, 2018-11-22:
I did extended testing with ds18b20 sensors on long cable, run into multiple issues. Most issues can be solved doing this.
Use twisted pair cable like cat5
Place all sensors on only 1 cable (don't wire a star)
Place the pullup resistance on the most remote sensor

peteeee, 2018-11-25:
Yes it worked fine with resistor at terminal end on SonOff basic (3.3VDC, GPIO14, Ground) star connected to 4 DS18B20 sensors connected via 22/4 alarm cable 1-3 meters in length using waterproof DS18B20 cables, raw DS18B20 chips, AAG 18B20 boards, Midon 18B20 boards and Hobby Boards 18B20 boards.  

Here is a drawing of configuration and Tasmota console messages with 4 sensors connected all extended from Sonoff basic terminals in a star configuration.  

Using this configuration with Home Assistant.  I like Espurna firmware flexibility versus Tasmota firmware.

[image]

peteeee, 2018-11-26:
Left it on all day with 1 short and 2 extended DS18B20's.  Looking at it 3-4 hours later it now shows three sensors.  Debug showing some errors relating to DHT sensor which doesn't make sense to me.

[image]

Valcob, 2018-11-26:
@peteeee All you need to do is to comment the line with `#define DHT_SUPPORT         1`  ;) and reprogram the device

peteeee, 2018-11-27:
Thank you Valcob.

Yes; went to 4 extended DS18B20 sensors and it is working now with one 4.7K ohm resistor except for the DHT debug errors.

I have wired the terminals on the SonOff basic WiFi for two GPIO pins.  Can I redefine the firmware to use GPIO 14 for DS18B20's and GPIO XX for one DHT22 combo sensor?

Valcob, 2018-11-27:
@peteeee sure just find the line or add one to you configuration in hardware.h  `#define DHT_PIN             5 <- this one might be something else`  and then just reflash the device

peteeee, 2018-11-27:
Thank you Valcob.

Here had set up PlatformIO to work with Tasmota and just configured it to work with Espurna firmware.

Still learning.

Found this section:

//------------------------------------------------------------------------------
// Dallas OneWire temperature sensors
// Enable support by passing DALLAS_SUPPORT=1 build flag
//------------------------------------------------------------------------------

#ifndef DALLAS_SUPPORT
#define DALLAS_SUPPORT                  0
#endif

#ifndef DALLAS_PIN
#define DALLAS_PIN                      14
#endif

#define DALLAS_RESOLUTION               9           // Not used atm
#define DALLAS_READ_INTERVAL            2000        // Force sensor read & cache every 2 seconds

//------------------------------------------------------------------------------
// DHTXX temperature/humidity sensor
// Enable support by passing DHT_SUPPORT=1 build flag
//------------------------------------------------------------------------------

#ifndef DHT_SUPPORT
#define DHT_SUPPORT                     0
#endif

#ifndef DHT_PIN
#define DHT_PIN                         14
#endif

#ifndef DHT_TYPE
#define DHT_TYPE                        DHT_CHIP_DHT22
#endif

So here can I keep the GPIO dallas sensors to GPIO 14 and change the DHT sensors to GPIO4 or any other.  Here modded my SonOff WiFi basic adding outside terminals for multiple GPIO pins, 3.3VCC and 5.0VCC and ground for multiple sensors connectivity.

peteeee, 2018-11-27:
Keep closing thread by accident fat fingering it.

peteeee, 2018-11-27:
Got 5 DS18B20 sensors working now.  2 are local which are less than 2 meters and 3 are further away.

I put one 4.7k Ohm resistor at the terminal end of the SonOff basic Wifi device and one 4.7k Ohm resistor at the end run of the longest wire (50 feet or so).

[image]

Next looking to add two DHT22 combo sensors on different GPIO pins.

So modding firmware to work with wired terminals and multiple GPIO pins.

Valcob, 2018-11-27:
@peteeee OMG  stop adding temperature sensors )))) 
but I would say that there is no way to add more DTHs then only one that is a software limitation sadly but there is a hope that it will be removed some day :)

peteeee, 2018-11-28:
Yes here got addicted to 1-Wire sensors in the late 1990's and weather using an AAG Weather station.
Over the years expanded to multiple 1-wire hubs using Midon Temp05, Temp08's and Dallas serial hubs...then added 1-Wire Hobby Board sensors.  

First MQTT testing was relating to using an RPi / Node Red / OWFS / Dallas hub...works great today...and now wanting to downsize a bit and go to a hub and spoke topology using updated SonOffs.

Ideally incorporating the OWFS library would be a nice thing.

I did notice that with Tasmota firmware you can connect 3 DHT22's and did find a patch to update an old ino file:

Single wire sensor
The following single wire sensors are supported:
    DHT22 Temperature and Humidity - DHT22 (3) in group 1

I am good here now with 5 DS18B20 sensors (extended) connected.

[402188] [MQTT] Sending SonOff-1Wire-1/temperature/0 => 68.0 (PID 734)
[402191] [MQTT] Sending SonOff-1Wire-1/temperature/1 => 71.1 (PID 735)
[402193] [MQTT] Sending SonOff-1Wire-1/temperature/2 => 70.5 (PID 736)
[402199] [MQTT] Publish ACK for PID 734
[402202] [MQTT] Sending SonOff-1Wire-1/temperature/3 => 71.7 (PID 737)
[402209] [MQTT] Sending SonOff-1Wire-1/temperature/4 => 67.8 (PID 738)
[402214] [MQTT] Publish ACK for PID 735
[402222] [MQTT] Publish ACK for PID 736
[402368] [MQTT] Publish ACK for PID 737
[402370] [MQTT] Publish ACK for PID 738

and will close this thread.

Thank you for your help Valcob.

peteeee, 2018-11-28:
Officially closed this thread as I have fixed my multiple extended DS18B20 issue.  Great stuff here!!!!

closed 2018-11-28 by peteeee

JBSullivan, 2020-06-24:
Is this conversion also possible with Sonoff SV boards?