xoseperez/espurna #1362: Telnet requires password

stefmorp, 2018-11-22:
Thanks for the great software, I have been using it for almost one year with great satisfaction. Last week I updated the Sonoff TH firmware to 1.13.3 and discovered that now Telnet requires a password. Looking at the changelog, I discovered this is a new feature, enabled by default, that can be disabled. I would like to disable it. However, I have not been able to do so: cannot find a switch on the GUI nor a parameter in the configuration. The documentation still says that Telnet works without password (probably requires an update). Am I missing something? Thanks in advance for your help! Stefano

mcspr, 2018-11-22:
Yep, it is build flag right now. Should be setting instead, thanks for noticing!

sfilinto, 2018-11-30:
same observed in the core version. I attempted the 2-step update process, Flashed the espurna-1.13.3-espurna-core-1MB.bin and now unable to complete the update using terminal ( both telnet & http ) as it prompts me for a password.
any suggestions on how to get past the core without having to physically access the device ?

mcspr, 2018-12-01:
@sfilinto have you tried `fibonacci`? unless have you changed it to something else

I've added the fix for this + exception for espurna-core
today's https://github.com/mcspr/espurna-nightly-builder/releases will include it

sfilinto, 2018-12-01:
> @sfilinto have you tried `fibonacci`? unless you have changed it to something else
> 
> I've added the fix for this + exception for espurna-core
> today's https://github.com/mcspr/espurna-nightly-builder/releases will include it

I did try fibonacci besides my password, but no luck. it prompts for a password but accepts no further then 2 characters. 
Anyways flashing from core can be worked around with the espota.py tool where it accepts the password.

closed 2018-12-11 by xoseperez