xoseperez/espurna #1366: RF-Bridge / EFM8BB1 Firmware

olili, 2018-11-22:
I flashed current EFM8BB1 Firmware from Portisch ( https://github.com/Portisch/RF-Bridge-EFM8BB1 ) to my RF-Bridge with ESPURNA 1.13.3.
It is somehow not working as expected: the red RF LED is continuously flickering. "B0" mode is not working. 
Any hint's? 

If I take the old forked rhx version ( https://github.com/rhx/RF-Bridge-EFM8BB1 ) both symptoms are not there. But therefore it is very unstable wrt serial UART communication ESP8266/EFM8BB1. 

What is the difference between these 2 firmwares? 

Thx in advance, O.

mcspr, 2018-11-22:
Use the one from [Releases](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases)

The issue is known so I'd track the Portisch's repo for when it is fixed.

mcspr, 2018-12-12:
[Using latest firmware from here](https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/generic_bucket), everything works as expected.

Base mode serial messages are properly sent + ack'ed by the EFM.
RAW codes successfully sent out, LED distinctively blinking when sending.

closed 2018-12-12 by mcspr

olili, 2018-12-30:
Hii,

thx. Basically working.
But I'm facing an issue with B0-Mode. Until now no successful send message
possible. Any hints?

P.