xoseperez/espurna #1367: Reduce ws / settings lists size

mcspr, 2018-11-23:
Current settings use `keyN` layout for lists:
https://github.com/xoseperez/espurna/blob/d2d92687c1b205b5ce02d4d9024b91c872fbc219/code/espurna/settings.ino#L335-L337
Same situation for websockets api. For example, relays, where `key` part is duplicated N times relays:
https://github.com/xoseperez/espurna/blob/d2d92687c1b205b5ce02d4d9024b91c872fbc219/code/espurna/relay.ino#L598-L606

To be more memory friendly - what about mapping them as simple arrays?
e.g. settings - for 8 relays `relayBoot => "0,0,0,0,0,0,0,0"`, similar for websockets `relayConfig[0] = ...`.

This saves *a lot* not duplicating `key` for every entry. Less EEPROM used, much less data transferred for rfbridge-like boards, much less RAM used for JSON.

Valcob, 2018-11-27:
I would go for a struct as tasmota does, yes I understand that it's level 100 cryptic but the saved space can make it worth(cause all the keys are gone also we can use unions for relay states, pins, lights, their values and so on) this way the saved space is of an order of magnitude from what we are using now, but that would require a much in depth redesign of how configs are stored, loaded, updated

xoseperez, 2018-12-07:
Still, cryptic code or data is not a good idea. One thing is packing the info is sent via WS (with the benefit of less memory usage doing JSON stuff) as @mcspr suggests in the PR. and another one is changing the way the data is stored in settings... 

@mcspr, let's discuss this further. I like the PR so far but I would limit the scope to the WS communication here and open a different one for EEPROM storage if we decide to change that too.

mcspr, 2018-12-08:
ref #1387, will update soon and link back to this
+note: js already used this way of packing lists to send back, but server side only did it sometimes

Will add another issue tracking settings then
(and regarding cryptic settings: one can always program embedis to serialize the data under the hood and have input/output be whatever)

closed 2019-04-14 by stale[bot]