xoseperez/espurna #1371: Create CONTRIBUTING file

mcspr, 2018-11-24:
See https://help.github.com/articles/setting-guidelines-for-repository-contributors/
Wiki already contains https://github.com/xoseperez/espurna/wiki/PullRequests, adapting it here would make it visible in the repo UI.

xoseperez, 2018-12-21:
I have created a `contributing.md` page under .github (not sure this is the best place). Also added templates for issue reporting. Comments welcome.

mcspr, 2018-12-22:
It can be in the root directory: https://github.com/atom/atom/blob/master/CONTRIBUTING.md

Going to the atom/atom and trying to do things, CONTRIBUTING.md link is shown:
"New Issue" has it to the side (screenshot in the doc)
Going to the **Issues** or **Pull requests** tab shows it at the top
So it can have some short info about issues (labels, duplicates, stale), prs (what it is now) and documentation (per-device quirks, building in ide / pio i.e. link to the wiki pages)?

Another thing I noticed: when doing "New Issue" it also links to SUPPORT.md if it is present
["Looking for help? Check out the project’s instructions for getting support."](https://github.com/atom/atom/blob/master/SUPPORT.md)

xoseperez, 2019-02-11:
Moved to root folder and added a preliminary SUPPORT.md page.

closed 2019-02-11 by xoseperez