xoseperez/espurna #1372: Sonoff Pow v2.0 led

rafariossaa, 2018-11-25:
Hi,
  I own several sonoff basic and Sonoff Pow R2. I am using them with MQTT, so far so good. Thanks for the hard work!!!.
  Now, I just bought two Sonoff Pow more, but my surprise is that they are v2. The header only have 4 pins, I soldered a pin header and flashed espurna 1.13.3 from the binary file (
espurna-1.13.3-itead-sonoff-pow-r2.bin). 
  Then I configure them to connect my home wifi. Without any problems to this point.

  The issue is, that the blue led (that briefly blink once when power supplied) doesn't not work. My guess that in v2 they have changed the pin that control the led.

rafariossaa, 2018-11-27:
Hi @xoseperez,
  I have some experience with programing arduinos, so if you guide me maybe I can help debugging this.
  I was thinking to compile a customized espurna with leds in different pins until I find the right one. Do you think this approach is ok ?.

Valcob, 2018-11-27:
@rafariossaa just open the case and follow the trace then update the led pin in firmware and upload it. And don't hesitate to create a PR for an updated config for this device. Thanks in advance.

rafariossaa, 2018-12-02:
Hi @nrdmtt , @Valcob , @xoseperez 
  :man_facepalming: Yes, that was the problem. :disappointed: 
  Thank you very much.

closed 2018-12-04 by mcspr