xoseperez/espurna #1377: alexa discover devices

eduardocanton, 2018-11-27:
Hello,

I have used last espurna in a sonoff basic. The espurna works fine with mqtt. I activated the alexa integration in the control panel,. After , I say "alexa discover devices", but Alexa dont find nothing (and detected other alexa devices in my network).

Must I do something more?

I use echo v2 generation.

wildwiz, 2018-11-27:
I have successfully used amazon `echo dot v2` both with direct detection of espurna devices and with node-red plugin `alexa local`, but amazon `echo dot v3` does not recognize neither espurna nor the node-red plugin announcements on the LAN. 

For my understanding the `echo v2` and the `echo dot v3` seem to detect devices in a different way from `echo dot v2` and that makes them fail (I no longer have the details, but seems the amazon device calls the website of the manufacturer of the items the LAN discovery was meant for).

If you use node-red, you might want to try `Node-RED` Alexa plugin by Ben Hardill alongside his plugin for node-red `node-red-contrib-alexa-home-skill`.

wildwiz, 2018-11-27:
You might also want to read the issue #1268 (Amazon Echo doesn't see devices after update to 1.13.3 from 1.13.2)

eduardocanton, 2018-11-27:
I have download the firmware to 1.13.2 and now it´s working. thank you.

closed 2018-11-29 by mcspr