xoseperez/espurna #1378: GPIO0 no conmutation

eduardocanton, 2018-11-27:
Hello,

I would like change the button configuration to use with a "normal switch wall", connecting with two wires  this button to one swith, and then, if the status in button change, turn on the relay and if it change again, turn off the relay .... (not conmutator,....like a normal swith)

is it possible?

Valcob, 2018-11-28:
@eduardocanton If I got your point correctly you want the relay to turn on when you switch is on and to be off when the switch is off , if so then all you need to do is to set the button type as is in suggested code
```// BUTTON#_MODE: A mask of options (BUTTON_PUSHBUTTON and BUTTON_SWITCH cannot be together)
 //   - BUTTON_PUSHBUTTON: button event is fired when released
//   - BUTTON_SWITCH: button event is fired when pressed or released
//   - BUTTON_DEFAULT_HIGH: there is a pull up in place
//   - BUTTON_SET_PULLUP: set pullup by software

```

just find the line with `#define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH` and change it to `#define BUTTON1_MODE        BUTTON_SWITCH| BUTTON_DEFAULT_HIGH` for your configuration of course

eduardocanton, 2018-11-28:
i understand.... but...if I would like to do with the GPIO 14?, is it the same?

Valcob, 2018-11-28:
just change the `#define BUTTON1_PIN         0` to `#define BUTTON1_PIN         14`

closed 2019-02-04 by stale[bot]