xoseperez/espurna #1379: MQTT configuration (broker and port) is deleted...

jsponz, 2018-11-28:
In some circumstances (no idea), after powering a node with the latest versions, the default MQTT broker and default port (1883) are replaced with gateway IP and port 8883.

There is no configuration on my code with these settings (gateway IP and port 8883). 

Any idea? Any help?

mcspr, 2018-11-28:
Hello,

Does this happen in the web UI or configuration is  saved like that (.json backup / restore)?
Can you show the output of `get mqttServer mqttPort` in the Debug panel?

jsponz, 2018-11-29:
I see on the web UI the new MQTT broker and the new port. 

I have found out the reason, but it seems a bit weird for me. 

I have hardcoded a MQTT broker and a port. The broker is on the cloud. 

I have not touched neither MDNS_SERVER_SUPPORT nor MQTT_AUTOCONNECT -> default values, i.e.

MDNS_SERVER_SUPPORT = 1
MQTT_AUTOCONNECT = 1

If in the local network there is a discoverable (avahi / zeroconf/ bonjour) broker, the node takes the IP and port of the local broker, ignoring the hardcode values. This is not the expected behaviour IMHO because if I have set a given MQTT broker, no matter exists locally, I want to have the connection to the desired MQTT broker. 

The current behaviour could be risky in some situations...

mcspr, 2018-11-29:
How do you change the defaults?

Seemingly it only uses autodiscovery if either `mqttServer` setting or `MQTT_SERVER` define is "" (empty string)
https://github.com/xoseperez/espurna/blob/d2d92687c1b205b5ce02d4d9024b91c872fbc219/code/espurna/mdns.ino#L19-L26
https://github.com/xoseperez/espurna/blob/d2d92687c1b205b5ce02d4d9024b91c872fbc219/code/espurna/mqtt.ino#L746-L748

xoseperez, 2018-12-07:
I cannot reproduce this. I have hardcoded an MQTT broker (192.168.1.100 below) different from the one I have in my home server (192.168.1.11):

```
[006420] [MQTT] MQTT brokers found: 1
[006421] [MQTT] Broker at 192.168.1.11:1883
(...)
[006424] [MQTT] Connecting to broker at 192.168.1.100:1883
[006426] [MQTT] Client ID: ESPURNA-B93C93
[006429] [MQTT] QoS: 0
[006432] [MQTT] Retain flag: 1
[006434] [MQTT] Keepalive time: 300s
[006437] [MQTT] Will topic: ESPURNA-B93C93/status
[006450] [MQTT] Connected!
(...)
get mqttServer
[011138] > mqttServer =>
[011138] +OK
```

As you can see it discovers the broken at 192.168.1.11 but connects to the hardcoded one, and the `mqttServer` setting is not set in the EEPROM.

xoseperez, 2018-12-08:
No. Steps to reproduce:
* `MQTT_SERVER` set to `192.168.1.100` 
* `MQTT_ENABLED` set to 1
* Build and flash
* Open serial monitor to the device
* Type `factory.reset`, then `set ssid0 xxx` and `set pass0 xxx`
* Reset the board
* The board resets, connects to the WiFi, finds local broker but connects to the one at 192.168.1.100 as expected