xoseperez/espurna #1380: NtpClientLib issues / alternatives

mcspr, 2018-11-30:
Based on #682 discussion. Some possible options, not dependent on Core version:
- [ ] use `configTime()` from Core / lwip (possibly reuse now() / year() etc. with appropriate wrappers above libc gmtime())
- [ ] directly fix fork of NtpClient / import it's code here, per [gmag11 comment](https://github.com/xoseperez/espurna/issues/682#issuecomment-427324096) (no need to switch to AsyncUDP, just adjust internals. wip + problems described below need to be addressed)
- [ ] replace with adapted [ezTime](https://github.com/ropg/ezTime) fork (not usable directly as a lib right now, settings are hardcoded)

`configTime()` has offsets with core 2.3.0, proper time zones since 2.4.0, but having it configured on that level is still problematic (ref ntp.ino:`ntpLocal2UTC()`, `NtpClient::decodeNtpMessage()`). Instead, timezone support could be added outside of ntp client, based on the work done in #1295.
Timestamp can be kept as UTC and local time can be applied whenever it is appropriate
`ntpDST` / `ntpOffset` / `ntpRegion` can still be read but proper support should be using TZ variables / otherwise stringified timezone offsets array

While LWIP option is being the easiest to implement, sntp does not return much info besides timestamp being set initially. One reason to keep ntp client on Arduino level is to have access to the connection status / timeouts etc., although those issues are not really actionable.

<summary>*Some additional info* <br>tldr: packet validation is wrong, DST affects sync timestamps, DNS blocking</summary>
<details>
- No udp packet sanity check / kiss-o-death packet. Sync is treated as a success by checking udp packet length, which will not detect problems with packet itself. Also, quoting https://tools.ietf.org/html/rfc5905#page-76
> A kiss-o'-death packet is an NTP header with leap 0x3 (NOSYNC) and
> * stratum 16 (MAXSTRAT). It tells the receiver that something
> * drastic has happened, as revealed by the kiss code in the refid
> * field. The NTP header fields may or may not be correct.
- Multiple raised issues when it is crashing from either ws or mqtt (sys context inside async callbacks), due to TimeLib directly doing network request when now() is called.
- Faulty DNS will stall NTP.getTime() method until the DNS timeout comes
- .begin() immediately trying to sync time
- DST implementation is overriding received timestamp right in the udp packet, having no obvious way to receive it untouched (set daylight to false, sync, back to true?)
</details>

Misiu, 2018-12-04:
@mcspr as You wrote in https://github.com/gmag11/NtpClient/issues/60#issuecomment-420296062 there is a callback that can be used to block everything till we have correct date and time, so I think LWIP should be used. If something is missing we can create tickets in official repo https://github.com/esp8266/Arduino. Hopefully 2.5.0 will be released soon, so maybe missing features will be added there.
NtpClient is awesome, but thats just another dependency that can be replaced with build in functions.
What do You think?

mcspr, 2018-12-06:
Yep. It is even called for *every* successful sync. And it would sure be nice to handle this on system level.

General issue is new core / lwip2 requirement, as it was pointed out in #682, and size increase in general.
Plus, 2.5.0 now enables exceptions (can be disabled back right now, but may be required by the Core in the future), which even further increases size :/ tough life for 1MB boards.

mcspr, 2019-02-21:
@xoseperez ping again about https://github.com/xoseperez/NtpClient/pull/2
i'd rather it still points to your repo for now + easier upgrade way (just git pull)
additionally, what about compile time version check for it? that way it can at least warn arduino ide users about the need to upgrade. 0 time is a cause for a lot of grief when using scheduler :/

mcspr, 2020-01-31:
re: #2132 
General stuff is already there, main problems:
- TimeLib compatibility is all over the place with types. there is also my own misunderstanding of how to replicate minute() / hour() / etc. without depending on localtime and,or gmtime.
- If that is solved, scheduler needs adjustment with its `useUTC` and how it treats timestamp after it is enabled
- I *really* like RPN syntax for scheduler, but I am not yet sure how to replicate "Restore last schedule" feature

PR #2132 (2020-02-04): NTP: use sntp app from lwip on latest Cores, replace ntpclientlib
subj

fix #1380 (yay)
fix #1629

TODO:
- timebroker. can it be used as scheduler replacement + rpn? experimenting with $tick1m, $tick1h.
slightly complicated because of restore-last-schedule since rpn is generic and scheduler part would need to iterate over $tick rules. Can use a different rpn context and ruleset / settings, maybe.
- fix lib_deps forcibly linking ntpclientlib + timelib. if not possible, drop 2.3.0 from supported altogether. if it does work, just switch implementations as needed. webui can have `ntplegacyVisible` [... cut]
closed 2020-02-04 by mcspr