xoseperez/espurna #1381: Maximum number of schedules

fernando-p-jesus, 2018-11-30:
Hello,

Is there a maximum number of schedules? Currently this number is limited to 10 (SCHEDULER_MAX_SCHEDULES). Can this setting be changed safely to higher value?

Thanks!

mcspr, 2018-12-01:
Hi,

It might not work as expected (or at all) with web rn as it might send too much data at once.

But even if it does not display in Scheduler panel, you can try using raw settings keys to describe the schedules past 10 by using terminal connection (telnet/serial/web debug) and `set key value` / `get key`
(`get schEnabled0`, `get schAction1` etc.)
https://github.com/xoseperez/espurna/blob/de7ff7722c2c0eccd34ae0d7b08005fe38720b66/code/espurna/scheduler.ino#L74-L80

fernando-p-jesus, 2018-12-03:
Thank you @mcspr!

I did a few tests with SCHEDULER_MAX_SCHEDULES set to 32. It seamed all right at first but after I opened the schedule web interface things started to go unstable. From the interface point of view it worked ok, and I was able to set/edit/delete all the schedules, but then after a reset I was unable to connect again and I ended up having to reflash. Maybe it had to do with what you commented about the web interface (too much data)...

This leads me to another question. Are there any plans to add mqtt support for schedules (ability to set/query schedules from mqtt)?

I am new here, so I may have missed some previous conversation or documentation on this subject (I have done several queries, though). My apologies if I am asking something I should have found somewhere :-)

Thanks!

xoseperez, 2018-12-08:
It's not the first time this has been requested. There are no plans to add this feature soon but your opinion might help to decide about it. My current approach is that, if you are able to send MQTT messages to your devices to set the schedules, you can also send them to switch them on/off, so I don't see the point in doing it.

fernando-p-jesus, 2018-12-21:
Thank you @xoseperez 

The subject here is that I have a use case where the devices will not always be connected to the internet, so I was looking for a solution to program the schedules (while connected) with some automated script, and then leave the devices working autonomously (with no connection).

But I could probably accomplish what I need by using the telnet connection (I am relatively new here and didn't had the time yet to read about the telnet interface).

Thanks!

closed 2019-02-26 by stale[bot]