xoseperez/espurna #1388: WiFi SSID with space in name not working correctly via web interface

arjanmels, 2018-12-01:
Cannot set a SSID with a space in the name correctly via the web interface. Instead of a regular space (ascii 32), some sort of unicode space is stored. Setting SSIS via serial (set ssid0 xxx) does work correctly.

mcspr, 2018-12-02:
Hello,

What browser / platform are you using to input the ssid? Is it copy-paste or just keyboard?

arjanmels, 2018-12-02:
Chrome with keyboard input.

mcspr, 2018-12-02:
Can't reproduce with Windows Chrome using "basic" ENG US. and typing '1 2 3 4' into the Add Network / SSID
SSID field itself is a basic input=text, not really see how it would break. I'd check input settings for OS & browser.

and btw you can actually check the character in question. Launch web inspector via the rclick menu on the input element in question and do `$0.value.charCodeAt(...index...)` in console.

arjanmels, 2018-12-02:
$0.value.charCodeAt(...index...) indeed gives 32. After further trials I cannot reproduce anymore. (I have seen it twice on 2 H*01 = ESP8266 devices).

Apologies for the waisted time. I will put together a new report once I can reproduce systematically.

closed 2018-12-02 by arjanmels