xoseperez/espurna #1389: DHT Sensor on SONOFF POW, PIN1 never works

schweini, 2018-12-03:
I tried connecting some DHT21 and DHT11 sensors to my Sonoff POW's PIN1 (usually RX).
I am aware of the 110VAC ground issue - but these sensors are completely isolated.
As a precaution, I disabled all 'terminal' and 'telnet' options, but the DHT Sensor breakout boards (with included pull-up) simply refuse to work ("Timed out").

I have a Dallas one-wire DS18B20 Sensor running on another Sonoff POW running exactly the same Espurna version (1.13.0c), connected to the RX Pin1, and that works great.

Am I doing something wrong?

Valcob, 2018-12-03:
@schweini What is your config in hardware.h? maybe you can post some logs from console from admin tab?

schweini, 2018-12-05:
I'm so sorry - I got the pin numbers wrong - 
TX == GPIO1 == next to GND
RX == GPIO3 == next to VCC

I could swaer I looked that up. Maybe some diagram was bad...

closed 2018-12-05 by schweini