xoseperez/espurna #1401: Flashing espurna version for Sonoff POW R2 gives a Sonoff basic

kris969, 2018-12-07:
Hi,

Thirst of all, many thanks for such a hard quality work!

I have recently bought several Sonoff (basic, dual, pow r2, rf bridge) and some ESP8266 esp02.
I flashed Espurna on all of them without any issue, except on pow r2 : in this case the device name is "SONOFF_BASIC" so I think something is wrong. I both used compiled firmware available on the wiki and also the one I compiled myself with PlatformIO.

Facing this issue I checked the configuration files in sources, everything seems to be good.
I tried the last version (espurna-1.13.3-itead-sonoff-pow-r2.bin) and the previous one (espurna-1.13.2-itead-sonoff-pow-r2.bin) the result is the same.

I'm sure I'm doing something wrong, but so far I have no idea about what it could be.
So I expect some support from someone to fixed the problem.

Regards

xoseperez, 2018-12-08:
I just flashed the `espurna-1.13.3-itead-sonoff-pow-r2.bin` image and it is the right one. My guess is that you initially flashed a sonoff-basic image and now flash the new image fails somehow...

kris969, 2018-12-08:
Thanks for your message.
In the meantime I have posted a reply.
I have understood
The pow R2 need to be connected to a main 220V supply to work correctly.

Regards