xoseperez/espurna #1407: Increase number of switches in RF bridge

prof0021, 2018-12-09:
It is not a issue. It is a request. I like espurna and use it in many of my things. However I have seen that tasmota allows to support unlimited number of RF codes by direct resending of the RF codes to mqtt. So it is not needed to configure every RF code in RF bridge. It could be done for example in home assistant. Is it possible to have in also in espurna?

xoseperez, 2018-12-09:
You already can do that, simply send the codes to `<root_topic>/rfout`, like:

```
mosquitto_pub -t rfbridge/rfout -m 2C56010403F2440014
```

LukasAV, 2018-12-27:
How to send codes via the  HTTP API instead of MQTT? I could not find a documented method.

xoseperez, 2018-12-27:
There are no API entry points for the RF Bridge. That's a shame. I'm adding it to this milestone since most of the code is there, it's only a matter or wiring it.

LukasAV, 2018-12-27:
Thank you very much. I already looked into it but sadly this is above my programming knowledge. I will be looking forward to 1.13.4.

closed 2019-01-04 by xoseperez