xoseperez/espurna #1409: Allow to set env_default with environment variable

Niek, 2018-12-10:
Currently ``env_default`` is hardcoded to in platformio.ini: https://github.com/xoseperez/espurna/blob/433f089c00a890fbefef3f3c425ad80dd13b0115/code/platformio.ini#L2

What about making this configurable, for example ``${sysenv.ENV_DEFAULT}``? That would allow users to change the environment in VSCode's default tasks.json without changing the platformio.ini file.

xoseperez, 2018-12-10:
You can use the `-e` option

```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "type": "shell",
            "command": "platformio",
            "options": {
                "cwd": "${workspaceFolder}/code"
            },
            "args": [
                "run",
                "-e", "wemos-d1mini-relayshield",
                "-t", "upload"
            ],
            "problemMatcher": [
                "$platformio"
            ],
            "label": "PlatformIO: Wemos D1 Relayshield"
        }
    ]
}
```

xoseperez, 2018-12-10:
My priority is that ESPurna build system should work from the command line using PlatformIO. PlatformIO provides a way to build the different environments from the command line automatically and manages dependencies beautifully. I try not to enforce any specific IDE (although issues related to the Arduino IDE have priority since it's what a huge majority of users use).

Maybe someone can help on this or maybe you should ask in the PlatformIO forum. There is a recent PR in the VSCode project that will allow adding an input field to a custom task. Again, that would be outside the scope of the integration of PltaformIO with VSCode, but an interesting option nevertheless.

Niek, 2018-12-11:
I understand your point of view, and I agree it would be better if this would be fixed in the or PlatformIO integration in VSCode. Relevant issues:
https://github.com/platformio/platformio-vscode-ide/issues/116
https://github.com/platformio/platformio-core/issues/1967