xoseperez/espurna #1414: Incorrect Temp Readings on Sonoff-RF and DHT11

BuildTheRobots, 2018-12-10:
Just got a couple of Sonoff-RF units with the intention of wireless switching some room lights.

Setup was a little fiddly but mostly painless. 

I have created a custom Sonoff-RF image and added temp/humidity support, sadly the readings I am getting (`0.0°C, 3% humidity`) are bogus.

I've tried two known working DHT22's and also two Sonoff RF units and get the same results. I have tried using both GPIO 14 as it's broken out to the pin header on the RF units as well as GPIO3 (which is also RX).

I've done exactly the same on a Sonoff POW unit and it works as expected, confused why the results are so wrong on the RF.

I'm probably doing something stupid, but I'm not quite sure what as yet. 

`build_flags = ${common.build_flags_1m0m} -DITEAD_SONOFF_RF -DINFLUXDB_SUPPORT=1 -DHEARTBEAT_REPORT_RSSI=1 -DDHT_SUPPORT=1 -DDHT_PIN=14 -DDHT_TYPE="DHT_CHIP_DHT11" `

`

xoseperez, 2018-12-11:
I'm not sure. The default configuration for the SONOFF_BASIC_RF defined GPIO14 as a button, maybe that's interfering with the sensor. Try comment out that part in the `hardware.h` file:

https://github.com/xoseperez/espurna/blob/8b84f89a64aebb48cde213303b3e83808d3b1342/code/espurna/config/hardware.h#L289-L291

BuildTheRobots, 2018-12-11:
No change commenting out those lines in the hardware.h

I've also changed the pin from 14 to 3, which is the serial port RX line - this works on the POW unit but fails with the same incorrect readings on the RF.

To confirm, all i'm doing is connecting the 3 pins on the sensor to power, ground and either RX or GPIO14 on the 5 pin header on the RF.

If the sensor isn't connected (either data or power missing) then the Espurna status page says "timeout" rather than the reading, so I believe it's managing to talk to the sensor somewhat correctly.

I am a blundering novice and don't have a clue what I'm doing, so my "success" with the POW is more luck than judgement. I've got a couple of basic units somewhere, will test on those and report back.

BuildTheRobots, 2019-01-02:
Apologies, I'd thought I'd replied before Christmas.

The sensor in question is one of the AM2302 modules which only exposes Positive, Negative and Data and already has the pullup resistor as part of the board.

I've also tried adding a second pullup resistor but there's no change.

BuildTheRobots, 2019-03-03:
After causing some sparks a few weeks ago, I left this alone for a bit.

Coming back to it last week, I re-built my POW image only to have the
sensors fail in the same way, so I'm now pretty confident this is something
in software, though I have no idea what I changed to make it work
temporarily.

Sadly, after letting platform io do an update I now can't build anything
hitting grails errors on windows. I'll attempt to get things working and
will report back if I ever work out what it is.
With thanks :)

closed 2019-05-09 by stale[bot]