xoseperez/espurna #1416: Read LED status

etatus, 2018-12-11:
Many smart plugs have a LED you can turn on/off via MQTT topic <base_topic>/led/0/set. However I can't find a way to read the current status of the LED (if it is turned on/off). The logical path should be <base_topic>/led/0 but it doesn't work.

Any ideas?

xoseperez, 2018-12-11:
That feature is not implemented. Since it only makes sense for MQTT-driven LEDs the status is the same as the latest /set command.

etatus, 2018-12-11:
If it is not complicated to implement, I think it would be an interesting enhancement, especially if you use toggle command instead set on/off.

etatus, 2018-12-14:
@xoseperez Espurna allows to keep previous state of switches on boot, but what happens with led states? It would be nice to have the same possibility.

xoseperez, 2018-12-18:
I flagged it as an enhancement but it has low priority from my point of view. Maybe you can suggest the code changes or do a PR.