xoseperez/espurna #1417: alexa not detecting sonoff S20, neither at 1.13.2 nor at 1.13.3

mamama1, 2018-12-11:
Hi

I have just flashed my S20 with espurna 1.13.2 but alexa fails to detect the plug.
In the alexa app, i go to smart home, the + symbol in the upper right corner, add device, plug, other and then start the search.

1.13.3 doesn't work either.

i do have a sonos one (which has alexa integrated), two fire TVs and the alexa app on my phone (android). an echo dot (3rd gen) is on it's way.

i have read that tasmota as well is struggling connecting with alexa currently and that they have switched to yeti (which's skill is not available in my country, austria *sigh*).

is there a solution to this, has anyone managed to connect to alexa?

thanks!

mamama1, 2018-12-11:
There seems to be a solution here, for v3 Alexa. Haven't tried it yet. Will this be implemented into espurna?
https://bitbucket.org/xoseperez/fauxmoesp/issues/66/fauxmo-with-echo-dot-3

Thanks!

mcspr, 2018-12-12:
Yep, I'd track the fauxmo repo for more detailed info. Hopefully to be resolved soon.

And closing as duplicate of https://github.com/xoseperez/espurna/issues/1268

closed 2018-12-12 by mcspr

mamama1, 2018-12-12:
not really a duplicate, as in my case it doesn't even work with espurna 1.13.2.
guess that's because of alexa v3...

mamama1, 2018-12-12:
ok i now installed my first echo dot, 3rd gen and BOOM belkin wemo emulation works instantaniously.