xoseperez/espurna #1419: Scheduler

kris969, 2018-12-11:
I am using the scheduler on a Sonoff DUAL to turn on an equipment from hours (2AM to 6AM).
I have set 4 schedules, the first two ones are enable the last two are actually disabled, I will use it later.
Unfortunately The schedule seems not to start at 2AM, I was awake at this later time and have a check. So I turn it on manually.
I will procced some tests around this in the next days, to try to reproduce this issue, if so I will come back to you.

Working on this, I had the following Idea.
I think it could be interesting, to replace setting of independent schedule with the setting of sequence, each sequence will be a start time and an ending time. During the sequence the relay could ON or OFF depending of needs. With such approach, it could be possible to periodically check if the GPIO is in the right state regarding to an available sequence, and to activate it again in case nonsense.

What do you think about this?

Regards

xoseperez, 2018-12-12:
Sure that's an idea and it would fit into a more versatile schedule. One that could take into account different inputs like relay status, time of day (of course), sensor values,...

kris969, 2018-12-12:
I could propose you to participate in writing code. 
it could be interesting to start in writing a little document to specify what we expect to do, in reality it seems we have started in that way.

I'm not sure to be efficient on the GUI, but I could work on the scheduler itself.

Do you agree?

kris969, 2018-12-13:
Hi,
A new discussion...
Of course it is a question of point of view.

Don't you think it could be nonsense to use scheduler when using MQTT on Epurna?
I'm thinking about this because I started for one of my devices with Epurna Scheduler, then afet having deployed a new Mosquitto broker somewhere in the Internet, I switched using MQTT on it, and...

I forgot to clear the local Epurna Scheluder. Of course the scheduler and MQTT actions were working together! It takes me few minutes to understand I got a conflicting settings.

What do you think about this?

xoseperez, 2018-12-13:
The scheduler works in parallel to any other means of switching the relays, that includes MQTT, the REST API or the physical button. The initial goal was to provide different options for the user (some will do MQTT, other prefer to have a standalone device (hence using the schedule).

mcspr, 2021-10-03:
fwiw, current version *does* scan all declared schedules for the specified action target-id and launches the last one
but, only when there's a schedule restore setting on
https://github.com/xoseperez/espurna/blob/e22f67e5d61ce8c3c1eb1f9a50f2d6261b8b8d57/code/espurna/scheduler.cpp#L65
https://github.com/xoseperez/espurna/commit/e22f67e5d61ce8c3c1eb1f9a50f2d6261b8b8d57#diff-e8b9a5d04af16462e032b4ceb5900c29571415364a39b566b9ca03ac55749fcf

mcspr, 2024-06-16:
Closing, ref. #2417

closed 2024-06-16 by mcspr