xoseperez/espurna #1422: Problem relay pulse

MaN093, 2018-12-12:
Hello,
I have a problem with the relay pulse mode i think.

For the connections, I followed this one : https://lofurol.fr/joomla/images/domotique/Sonoff_Dual_R2_Volet-roulant.png
For configuration : https://imagizer.imageshack.com/v2/1003x901q90/922/WZGoaS.png

My problem : I have to stay on the button otherwise the shutter stops. I just made some tests by changing the puise mode and nothing changes. However, surprisingly, I can lift the shutter or down by pressing 3 times on the switch (one time: the shutter goes down and stops immediately, two times: the shutter goes down and stops immediately again and three times : nothing happens for 1 second and it goes down completely).

Below the debug log with this test:

-----------

[841827] [BUTTON] Button #0 event 2
[841831] [RELAY] #0 scheduled ON in 0 ms
[841842] [RELAY] #0 set to ON
[841844] [RELAY] Scheduling relay #0 back in 28000ms (pulse)
[841979] [BUTTON] Button #0 event 2
[841982] [RELAY] #0 scheduled OFF in 0 ms
[841993] [RELAY] #0 set to OFF
[842760] [BUTTON] Button #0 event 2
[842762] [RELAY] #0 scheduled ON in 0 ms
[842774] [RELAY] #0 set to ON
[842776] [RELAY] Scheduling relay #0 back in 28000ms (pulse)
[842920] [BUTTON] Button #0 event 2
[842922] [RELAY] #0 scheduled OFF in 0 ms
[842934] [RELAY] #0 set to OFF
[843813] [BUTTON] Button #0 event 2
[843816] [RELAY] #0 scheduled ON in 1015 ms
[843937] [RELAY] Setting relay mask: 0
[843959] [BUTTON] Button #0 event 2
[843961] [RELAY] #0 scheduled ON in 869 ms
[844833] [RELAY] #0 set to ON
[844835] [RELAY] Scheduling relay #0 back in 28000ms (pulse)
[845840] [RELAY] Setting relay mask: 1
[853290] [MQTT] Connecting to broker at 192.168.0.21:1883
[853293] [MQTT] Client ID: ESPURNA-9D7208
[853294] [MQTT] QoS: 0
[853295] [MQTT] Retain flag: 1
[853296] [MQTT] Keepalive time: 30s
[853297] [MQTT] Will topic: Volet Angeline/status
[859421] [MQTT] TCP Disconnected
[859424] [MQTT] Disconnected!
[872839] [RELAY] #0 scheduled OFF in 0 ms
[872845] [RELAY] #0 set to OFF
[873847] [RELAY] Setting relay mask: 0

----------

Thanks for help.

mcspr, 2018-12-12:
Hello,

This might be related to https://github.com/xoseperez/espurna/issues/1254#issuecomment-442585302 ?

If you can build the firmware yourself, can you try adding this lines:
```cpp
  pinMode(9, INPUT);
  pinMode(10, INPUT);
```

Inside setup() here?
https://github.com/xoseperez/espurna/blob/d2e0b12189c19b39a7dc17b2e8bd428c309e14d2/code/espurna/button.ino#L141-L142

xoseperez, 2018-12-13:
It looks like button 0 is only reporting event type 2, which means it is probably configured as a switch. @MaN093  Did you change the device configuration in `hardware.h`?

MaN093, 2018-12-13:
@xoseperez i didn't change the device configuration. I'm novice for instance =)
@mcspr i will try this method with tutorial i think this week end

Misiu, 2018-12-17:
@MaN093, as You have Sonoff Dual R2, could You try changes described [in this issue](https://github.com/xoseperez/espurna/issues/1254#issuecomment-442585302)? I'd like to buy couple of Sonoff's and use them as roller shutter.

closed 2019-02-22 by stale[bot]