xoseperez/espurna #1429: Blitzwolf BW-SHP6 support

m4rcu5, 2018-12-15:
Hi,

It would be nice to have the BW-SHP6 supported. As I have a pair of them, as well as the SHP2, I was preparing a full-on tear-down and comparison of both, until I stumbled across this issue: https://github.com/arendst/Sonoff-Tasmota/issues/4091.
It mentions that both devices are mostly the same, pin compatible and therefore the firmware of the SHP2 should work.

I've tested the Espurna SHP2 firmware and it seems to be fully compatible. I could not detect any issues;
LEDs and switching works as expected, power consumption seems right (I have no validated load to test with at the moment)

I did make a bunch of photos which I would like to add to the wiki, including instructions on how to disassemble and flash the firmware. If one of the devs can confirm and add a SHP6 build, I will update the wiki.

mcspr, 2018-12-18:
Note on the wiki would be welcome.

Question: is the any need to add separate binary if SHP2 .bin works?

m4rcu5, 2018-12-18:
> Note on the wiki would be welcome.
Let me fix that in the next days. I already have the photos ready to roll. :smile: 

> Question: is the any need to add separate binary if SHP2 .bin works?
If there are indeed no other changes, then the main reason would be to have the correct model in the interface?

mcspr, 2018-12-18:
Would've liked to replace this, but DEVICE is used in several other places and would look wonky :/
https://github.com/xoseperez/espurna/blob/59685150b4ea6358253a24c4ba6792cf2e695131/code/espurna/config/hardware.h#L2661-L2662
[image]

Well, might as well add duplicate hardware definition / platformio.ini config + notice.

xoseperez, 2018-12-18:
Maybe we should decouple the DEVICE from the UI strings...

mcspr, 2018-12-20:
Now that I think about it - just adding a check for SHP define to set `boardName` setting like on the screenshot above. I was under assumption it was used directly, but everything goes through `set/getBoardName` / `DEVICE_NAME` value.

evlo, 2019-01-20:
I read these are quite unreliable regarding power consumption measurement. How bad is it if I expect 70-100% load 24/7?

miljume, 2019-02-18:
I have flashed my SHP6 with espurna-1.13.3-blitzwolf-bwshp2.bin and it flashes just fine. After a reboot (still connected to the FTDI and the computer) it fires up the ESPURNA-XXXXXX AP and I can connect and configure the plug. But when I re-mount the PCB in the socket and plugs it into a 230V outlet the Wifi fails to connect! It neither connects to the configured SSID or fires up the AP. I have tried refreshing it several times and also tried Tasmota but with the same issue. Anyone experienced the same issue?

xoseperez, 2019-02-21:
Does it only happen when connected to mains? What happens if you reset the device while connected to your computer? It may be that the memory chip is faulty...

afrihagen, 2019-02-24:
https://github.com/ct-Open-Source/tuya-convert
How to do it wirelessly in the first place....
must admit that I have some problems in making this work:
Mint Linux  on a HP laptop (fresh install) I do not get vtrust-flash up and running
Raspberry pi b+ v 1.2 Raspbian (fresh install) I get vtrust running, but the script do not find the SHP6.

Anders F

victorclaessen, 2019-02-24:
Hi, the SHP2 firmware works on my SHP6 as well. Great!

Do any of you also have the idea that the red led should be inverted from it's current configuration? Right now the red led is off when the socket is live (i.e. output is enabled), and the red led is off when the socket is off. Since most people associate red with danger, wouldn't it make more sense for the red led to be on when the socket is live?

Vic

xoseperez, 2019-02-24:
By default it is defined as in "Find me" mode, which means the LED is on when the light is off (to be able to find it in the dark).

victorclaessen, 2019-02-24:
Yes, that is a really useful feature! And probably fine as the default. But I think I should elaborate:

I do not have the SHP2, and I guess I should have mentioned that. So I do not know what the led layout is on the SHP2. 

On the SHP6, there are two leds: a blue led, and a red led. When the SHP6 is flashed with the espurna SHP2 firmware, the function of the blue led can be set with 'led mode' on the 'general' page. The red one seems to follow the relay, but inverted (so off when the relay is on, and vice versa). It's the function of the red led that I think would make more sense when it wouldn't be inverted (i.e. in my opinion it would be better if the red led was on when the relay was on). And the function of the red led cannot be set from the web interface right now.

victorclaessen, 2019-02-25:
Ah, regarding the above; I just realised you meant "when the light _that is switched by the relay_ is off". Previously I assumed "when the light is off" to mean "when the sun has set" :-D
I never used these to switch lamps, only washing machines etcetera.

miljume, 2019-02-27:
> Does it only happen when connected to mains? What happens if you reset the device while connected to your computer? It may be that the memory chip is faulty...

Yes, it only happens when I connect the device to mains. The chip flashes OK and when I restart I can change password and connect again (still connected to computer). I add my network in the config and then disconnect from computer and re-assemble the unit, put it in the main socket and then NOTHING. The led is solid RED and I can't find any unit on the network (I use DHCP). Can the unit be faulty even though it flashes without errors and functioning well while connected to computer?

miljume, 2019-03-02:
> Yes, it only happens when I connect the device to mains. The chip flashes OK and when I restart I can change password and connect again (still connected to computer). I add my network in the config [...]

Issue solved!
It seems like my Esptool installation was faulty. Re-installed it on another computer and then it worked directly. Strange though that no errors occurred during flash and that the firmware was partly working. But from now on I will use Tuya-convert to avoid opening the plug, worked like a charm the first time!

xoseperez, 2019-03-25:
Working now...

[image]

closed 2019-03-25 by xoseperez

victorclaessen, 2019-03-25:
Awesome, thanks!