xoseperez/espurna #1431: TECKIN SP22  (ASIN B07D5V139R) - fw upgrade  pins and pictures

juanmahl, 2018-12-16:
Here's how to connect the Teckin SP22, variant of the Blitzwolf SHP2, for fw flashing.

This is related to other issues https://github.com/xoseperez/espurna/issues/1283

As the case is glued (no screws anymore) it's really difficult to open. This first case  was a nightmare. Hope for the next ones the cases will be less damaged. Smd air soldering iron (or heatgun) are your best friends in this case.

The device is based on TYWE2S  (https://fccid.io/2ANDL-TYWE2S/User-Manual/Users-Manual-3596121)
So according to the pinout the pins are located in a row: VCC (3.3v), GND, RX, TX.
Using a 3.3V usb serial adapter connect to the pins (tx swaps with rx)

Red - VCC
Blue- GND
Purple - RX
Green - TX

[image]
[image]

Also to make the ESP go into flash mode, you have to connect DI0 to GND. That's a bit tricky as DI0 is not available in the pionout but in the PCB
[image]

There's no binary available for the model yet but it's already in the code as teckin-sp22-v14 and teckin-sp22-v14-ota, so it can be compiled by cloning git and making changes in platformio.ini

You can have more info (and better pictures) in https://github.com/arendst/Sonoff-Tasmota/issues/3950

mcspr, 2018-12-18:
If you do not mind, this can be converted into a wiki page
Separate from original SHP2, as I understand this is for incompatible v1.4 board?

Minor note: prebuilt binaries from `dev` branch are already building "nightly" [here](https://github.com/mcspr/espurna-nightly-builder/releases)

xoseperez, 2018-12-18:
Yes, teckin-sp22-v14 image is valid for this device. Even if they are labeled as v1.6 in the PCB.

juanmahl, 2019-01-02:
> If you do not mind, this can be converted into a wiki page
> Separate from original SHP2, as I understand this is for incompatible v1.4 board?
> 
No problem in espurna team using these pictures and text  to update the wiki

closed 2019-01-02 by juanmahl

Wikibear, 2019-01-05:
Thanks for your pictures... I totally forgot to take one. TX RX must be connected directly but GND and power you can also connect here, which is easier to solder:

[image]

mcspr, 2019-01-15:
Added to the wiki (+comment above)
https://github.com/xoseperez/espurna/wiki/Hardware-Teckin-SP22-v1.4-and-v1.6

huelvayork, 2019-02-09:
The method described here works fine for flashing espurna on TECKIN SP22:

[https://github.com/ct-Open-Source/tuya-convert](https://github.com/ct-Open-Source/tuya-convert)

No need to open the case.

darkoddio, 2019-02-22:
> The method described here works fine for flashing espurna on TECKIN SP22:
> https://github.com/ct-Open-Source/tuya-convert

hi all
I followed this method and initially flashed the espurna-1.13.3-blitzwolf-bwshp2.bin image and obviously I discovered that neither button nor rele were working.
Then I found this issue here, downloaded the nightly build and everything seems working perfectly now.
I was just wondering when the final release of the Teckin SP22 image will be published :)
If there is something to test on this plug just let me know

baldfox, 2019-09-21:
For those of you looking to flash teckin SP22, here's what I did to get mine working with OTA convert 2!
some pointers:
1) to put it in pairing mode, i plugged it in (the light was off) and I held it down for about 6 seconds... the light was blinking blue and off, not red and blue (I tried many times with red and blue flashing but it wouldn't work for some reason).

2) I followed this video - https://www.youtube.com/watch?v=6_IZEa9JcC8
I also manually installed libssl-dev

3) I left my original tuya convert folder and just opened a new folder on the raspberry pi...

4) I cloned this repository into it:
https://github.com/M4dmartig4n/tuya-convert

Follow the instructions in the vid...

5) The lights went out on the teckin SP22 and i thought it had messed up, but it just worked. You need to be patient.

Then I proceeded to flash using the curl commands...
Hopefully this helps, it's not the cleanest instructions, but it worked for me. My bricked SP22 is now in Home Assistant :)

xefil, 2019-11-26:
Hello @baldfox !
Sorry, I've got fresh new SP22. Let me understand.
It's possible to flash it without open it?
Thanks!
Simon

baldfox, 2019-11-26:
hey. Well my previously bricked devices were flashed using tuya convert 2.0.
I think you will be able to do it. Do NOT connect the devices to wifi / the tuya app first. Just go straight for the flash. I never needed to open mine. I'm even consisdering buying more.

xefil, 2019-11-26:
Hi @baldfox , wow, fast answer :-)
Damn, I was right now trying to open it with any tool I've around the office, without success :-D
Ok, also I need to follow the video 'https://www.youtube.com/watch?v=6_IZEa9JcC8' Uh, long video.
I'll give it a try, thanks!
Simon

xefil, 2019-11-28:
Hello @baldfox !
Well, no god news. I think I'm not able to put it in pairing mode correctly, or, I can but the flash wont work. If I press the button for some seconds, it starts to blink fast, but between blue and red. Then, I can see it connects to the AP and starts to do some stuff, but it appears something like
"WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware"
Something wrong or maybe something changed? The device was not updated first.
Thanks, Simon

mcspr, 2019-11-28:
Perhaps related to the https://github.com/xoseperez/espurna/issues/1912 & https://github.com/ct-Open-Source/tuya-convert/issues/310?
Those are Realtek chips, so the warning might be right

xefil, 2019-11-29:
Ok, thanks @mcspr , so no way to deal with it :-(