xoseperez/espurna #1432: Activate toggle on code

emsilva, 2018-12-16:
I've added a keypad on my wemos d1 mini so that the relay can be activated by a password. How would I toggle the mqtt state through code? I want to use the internal API to do that instead of creating a new one. Is there a way to simulate a button push through code?

Thanks

mcspr, 2018-12-17:
If by toggling state you mean sending current relay state, `relayMQTT` sends it out:
https://github.com/xoseperez/espurna/blob/59685150b4ea6358253a24c4ba6792cf2e695131/code/espurna/relay.ino#L809

For buttons see this event handler (and parts related to the Sonoff Dual, without the gpio watching part):
https://github.com/xoseperez/espurna/blob/59685150b4ea6358253a24c4ba6792cf2e695131/code/espurna/button.ino#L95

emsilva, 2018-12-24:
I'm sorry, what I mean i toggling the relay state. Like for example...
if its 
[RELAY] #0 set to OFF
I want to know what is the function I can call to toggle it to 
[RELAY] #0 set to ON

so I can use it for then the password is typed correctlyl. Thanks

emsilva, 2018-12-24:
I found it!

`relayToggle(0);`

closed 2018-12-24 by emsilva

kappafix, 2020-03-08:
Hi @emsilva, I would like to add a keypad to my esp and espurna, do you have some code to share or suggestions on how to do it? My target is write the pin/password on the keypad and then send the code by mqtt on the server, that's it.
Could you help me?