xoseperez/espurna #1433: no timestamp in json flow

kris969, 2018-12-16:
Hello,

Here is my request about something I don't really understand yet, I'm not sure it's an issue...
I'm using a DHT22 sensor, I tuned Espurna in MQTT parameters to Use JSON payload. It iis writen ;
All messages (except the device status) will be included in a JSON payload along with the timestamp and hostname and sent under the <root>/data topic.

I will have a look on the code, to try to understand why I don't get the timestamp.

Unfortunately, it's seems no timestamp is there!
Once tit will be correct, the next step for me will be to store the flow to inFluxDB, so It will be nice to have the timestamp to do this.

Here after the message I receive from debug log using red-NODE:

object
topic: "ESPURNA-DHT22/data"
payload: "{"temperature":6.1,"humidity":99,"mac":"80:7D:3A:75:18:14","host":"ESPURNA-DHT22","ip":"192.168.1.17","id":6009}"
qos: 0
retain: false
_topic: "ESPURNA-DHT22/data"
_msgid: "c1f345aa.620208"

mcspr, 2018-12-17:
Cannot verify this with current version:
```
[210464] [MQTT] Sending test/status => 1 (PID 1)
[210568] [MQTT] Sending test/data => {"app":"ESPURNA","version":"1.13.4-dev","board":"NODEMCU_LOLIN","host":"ESPURNA-3601AF","ssid":"<ssid>","ip":"<ip>","mac":"<mac>","rssi":-63,"uptime":1210,"datetime":"2018-12-17 02:29:59","freeheap":23728,"relay/0":1,"vcc":2777,"loadavg":1,"time":"2018-12-17 02:29:59","id":6} (PID 1)
```
Does it show time on Web NTP panel?

kris969, 2018-12-17:
Hi,

Thanks for your help :)
No, it doesn't!
It could be the reason why,  I suppose.

Something is curious with NTP : actually I'm using 8 devices, all of them running the same Espurna version, and with the same NTP settings. I notice only TP of them are synchronized. I don't know the reason why. I know that when the current device time is to far away from NTP server time, NTP client could decide to do nothing. But it should not be the case at startup when I force a reboot.
I will have a look on this next evening, trying to poll other NTP server than the default settings one.

Regads

kris969, 2018-12-18:
Hi,

I have got it, it works:
Based on the information given by __mcspr__, I focus on NTP.
I have found discussion about ntp client issues, and some workaround base on timeout tuning.
My system is based on a raspi runing node-RED, mosquitto, inFluxDB, and several Espurna nodes (about 10 today, it will increase in the next days), all of this staff is on my LAN.
I am aware about NTP, cause I worked on it many years ago. Without consideration regarding the development stability of the NTP Client lib used with Espurna, it's somewhere a nonsense to have many nodes requesting ntp pool server, cause with the rising up number of IOT the overload of ntp servers becomes a reality.
My solution is very simple, I have installed an ntp server service on my raspi and changed the ntp settings for all my nodes to point to my new ntp server that is now my raspi. In my case the raspi have of course a static IP that is 192.168.1.36, so I put that IP in ntp Espurna settings, and every nodes are sync'd ! Great !  

To install ntp service on raspi, it's really simple: __sudo apt-get install ntp__
That's all!
It is possible to change the ntp pool server in /etc/ntp.conf. I didn't change anything, debian ntp pool is use by default.

I have a look on MQTT json using debug on my node-RED, the date and time are there. Great !

Now that it's working, having a deeper look on Espurna I have the following suggestions.
--When we plan to store sensor data to inFluxDB (or any other database) using json format, we do need the timestamp. But it's impossible to know if node is sync'd or not. Don't you think it will be usable to have this information provided in status?
--When setting the timezone in the ntp section, positive values are on the top of the list and negative ones on the bottom. I'm maybe a special guy but it seems for me non natural. several time I did it quickly and set GMT-1 rather than GMT+1, that is the good one for me.

Regards  ;)

mcspr, 2018-12-18:
Nice. Setting local NTP is best here as it is also avoids the "internet" part. Closing as resolved.

Regarding the second part:
> --When we plan to store sensor data to inFluxDB (or any other database) using json format, we do need the timestamp. But it's impossible to know if node is sync'd or not. Don't you think it will be usable to have this information provided in status?

Timestamp is already known - it is when message is received. I do not use node-red personally, but I am sure it has some form of templating to add timestamp manually somewhere in the processing.
Also note that influx module does not send timestamp. [Citing docs (near the end)](https://docs.influxdata.com/influxdb/v1.7/guides/writing_data/#writing-data-using-the-http-api):
> If you do not specify a timestamp InfluxDB uses the server’s local nanosecond timestamp in Unix epoch. Anything that has to do with time in InfluxDB is always UTC.

<br>

> --When setting the timezone in the ntp section, positive values are on the bottom of the list and negative ones on the top. I'm maybe a special guy but it seems for me non natural. several time I did it quickly and set GMT-1 rather than GMT+1, that is the good one for me.

IMO it is pretty common listing order :) e.g. reading this image left-to-right: https://en.wikipedia.org/wiki/List_of_UTC_time_offsets#/media/File:Standard_World_Time_Zones.png reading left-to-right
(wip "nicer" timezone support should solve this anyways)

closed 2018-12-18 by mcspr