xoseperez/espurna #1436: SPIFFS Size Negative.

arihantdaga, 2018-12-18:
#### ESP Arduino Core version - 2.5.0 (Just few commits ahead).
#### ESPURNA Version latest from git. 1.13.4-dev
#### Board - NodeMCU_LOLIN

I am seeing this log in terminal on startup. 

```

---8<-------

[000120] [MAIN] ESPURNA 1.13.4-dev
[000120] [MAIN] xose.perez@gmail.com
[000121] [MAIN] http://tinkerman.cat

[000121] [MAIN] CPU chip ID: 0x8A1BBE
[000124] [MAIN] CPU frequency: 80 MHz
[000127] [MAIN] SDK version: 3.0.0-dev(c0f7b44)
[000132] [MAIN] Core version: 3de0a4a7
[000135] [MAIN] Core revision: 0
[000137] 
[000139] [MAIN] Flash chip ID: 0x164020
[000142] [MAIN] Flash speed: 40000000 Hz
[000146] [MAIN] Flash mode: DIO
[000148] 
[000149] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[000156] [MAIN] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[000164] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000171] [MAIN] Firmware size       :   563328 bytes /  138 sectors (   1 to  138)
[000178] [MAIN] Max OTA size        :  3624960 bytes /  885 sectors ( 139 to 1023)
[000185] [MAIN] SPIFFS size         : 4294946816 bytes /   -4 sectors (1024 to 1019)
[000193] [MAIN] EEPROM size         :     4096 bytes /    1 sectors (1020 to 1020)
[000200] [MAIN] Reserved            :    16384 bytes /    4 sectors (1021 to 1024)
[000207] 
[000208] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[000212] [MAIN] EEPROM current: 1018
[000216] 
[000217] [MAIN] EEPROM:  4096 bytes initially |   219 bytes used ( 5) |  3877 bytes free (94)
[000226] [MAIN] Heap  : 40872 bytes initially |  5600 bytes used (13) | 35272 bytes free (86)
[000233] [MAIN] Stack :  4096 bytes initially |  1120 bytes used (27) |  2976 bytes free (72)
[000241] 
```

Notice the ` [MAIN] SPIFFS size  ` Line where SPIFSS Sector count is negative. Although this is not a major concern. My major concern is that yesterday 2 of my devices suddenly stoped working and on connecting them to serial i found that there was frequent WDT RESETS before printing anything. I tried dumping code again, but dumping any code would not work and the device keep restarting due to wdt reset. Even the simplest serial write code would not work. Until i upload the code using Clear "All Flash Contents" on the Arduino IDE. After that its starts working again. 
And i am not understanding the reason behind it, what could have possible corrupted. in the flash. 

I understand that this may not be an issue with the Espurna Exactly and i am sorry if this looks irrelavant, It can be an issue with the ESP Core as well. But suddenly my other devices are also reaching to the same thing. It may concern other people as well who are using latest ESP Core 2.5.0. 
Can somebody help me or point me in the right direction. 
**Its not booting, with any program. unless i clear all flash contents.**

mcspr, 2018-12-18:
Negative SPIFFS is because of this thing:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.4m.ld
Espurna does `end - start` to find out the size, which would work if they were correct
0x40600000 - 0x405FB000 = 20480 (as comment mentions)
0x405FB000 - 0x40600000 = 4294946816  because this actually becomes (2 ** 32) - 20480

On espurna side it should be more vigilant and also check for zero in _SPIFFS_page / _SPIFFS_block

Core ld files generator somehow mixed up the values (or that was manual edit), so that's a Core bug too.
[Some time in the 2.4.2...2.5.0 cycle this was changed from 3MB spiffs to 0](https://github.com/esp8266/Arduino/commit/cc0bfa04d401810ed3f5d7d01be6e88b9011997f#diff-54af750862005b1a7d50aab347930ae1)
[And back to 20KB spiffs with start and end mixed up](https://github.com/esp8266/Arduino/pull/5122/files#diff-54af750862005b1a7d50aab347930ae1)

I have not checked the code further, but is plausible that spiffs calculation is used for some other things which *can* break the last 16KiB

arihantdaga, 2018-12-19:
@mcspr Thank you for this. 
I dug into the issue. It happened when the device downloaded OTA from server. I recreated the issue multiple times and verified it was due to OTA(After OTA, devices would not work due to bad last sectors.). Then i updated the core to the latest(Including your commit) and it has fixed the issue. Now OTA is working fine. 
I think we can close the issue, its fixed with the latest core.

closed 2018-12-19 by arihantdaga