xoseperez/espurna #1437: Add support for "MagicHome Led SPI Controller" [with pictures]

TheCrypt0, 2018-12-18:
_**Preamble:** This is my first "post" on GitHub so I apologize in advance if something will be missing or misformatted, just tell me and I'll edit the message._

Hey everyone! I recently received from amazon this controller in a bundle with 5M of 5050 LEDs (controlled by a WS2811 every 3 leds) and a power supply. It was really cheap, something like 16€ for the full bundle and I started playing with it.

[image]

After setting it up with the MagicHome app I was greeted with this dialog (which shows that there are plenty of LED chips supported):

[image]

And then I chose the correct chip (which apprently is wrongly displayed because the IC on the led strip is the WS2811 but, when selected, it just wouldn't work so I had to choose the WS2812B). Here's a couple of screenshots of the controller app:

[image]

### The PCB

Even if the controller is labeled as SPI I'm not sure if it's actually as it says. Anyway, here are some pics of the PCB, it appears that in addition to the main ESP8285MOD there are a couple of ICs (one labeled ARM).

[image]
[image]

### ICs list
1. AMS1117 3V3 voltage regulator
2. GBAAD (a mosfet maybe? Cannot find any datasheet)
3. GD32F130F8P6 (ARM Cortex-M3 RISC - [Generic Datasheet](https://datasheet.lcsc.com/szlcsc/GigaDevice-Semicon-Beijing-GD32F130K6U6_C81552.pdf))
4. HC245 (Octal bus transceiver - [Datasheet Link](https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf))

### Summary
It would be really cool to support this module. However we don't know how the ARM processor communicates with the ESP yet. Further analysis is needed.

Do you have any suggestion?

lobradov, 2018-12-18:
Can you provide few additional pictures that show traces a bit better?

I'm trying to figure out if ESP is used as 'modem only' for enabling ARM to communicate via WiFi (in which case, espurna can't help) or ESP is driving ARM for whatever purpose... 

//_ on phone

TheCrypt0, 2018-12-18:
@lobradov yes, sure. Thanks for the fast reply! 

The closeup inspection apprently shows that the TX and RX lines from the ESP go directly to the ARM IC. Maybe an hardware modification (a cut to the traces from the ARM and and couple of wires from the ESP) to route a GPIO directly to the data out (or to the transceiver ic) would allow espurna to be usable on this module.

 
[image]
[image]
[image]

TheCrypt0, 2018-12-27:
Hi @ricolk, you are correct. The TX/RX lines go directly to pin 17 and 18 (PA9 and PA10) and the buffer is connected to pin 9 and 10 (PA3 and PA4).

I cannot probe the TX/RX lines right now, btw I noticed that they are connected to the ARM through a 0 ohm resistor, so disconnecting them should be pretty easy (I should have a hot air soldering station lying around). The buffer is connected to the ARM via 2 30 ohms resistors, so fortunately there will be no need to manually cut traces.

Tomorrow I'll do a dump of the original firmware, disconnect and (maybe) de-solder the ARM processor and create a connection with a wire (and a 1k resistor, just to be safe) between an ESP gpio and the pin 4 (A2) of the buffer (the one that controls the DOUT leds' data). However, even though if I've some experience with the ESP, espurna is new to me and I'm not sure where to look to start implementing a new device. 

Time for a couple of questions: 
- Is there a guide (or a procedure) to follow to add a new device to espurna? I couldn't find any in the wiki, haven't searched in other issues yet.
- Is the WS2812B protocol already present in the espurna firmware?

(Bonus: I've just bought a RGBW MagicHome smart bulb which I think will be super easy to add as a new device, just 4 mosfets connected to the GPIOs)

The effort that has been put into espurna is great and I'd happily contribute to the project!

dobryy, 2019-04-07:
@TheCrypt0 have you had any luck with this controller?

TheCrypt0, 2019-05-10:
Hi @dobryy,
I haven't got time to look into this controller yet, however I will look into it some day.

purplepiccolo, 2019-07-23:
Hey @TheCrypt0, I also had the issue of the 2811 option not working with the WS2811 pixels I have. However, when I tried WS2812B, I can seemingly only control the first 3 of the 50 pixels I had connected. for you the WS2812B option worked well for the WS2811 pixels you had? I wonder what my disconnect is.

brandond, 2019-12-07:
I have a similar device, just picked it up off Amazon. I'm going to see if I can grab a capture of the comms between the esp and arm MCUs with a logic analyzer. I'm hoping its a variant of the Tuya protocol.

[image]
[image]

purplepiccolo, 2019-12-07:
I picked mine up on alibaba. Does the amazon version work? interesting chip, wish it already did what I wanted it to do.
[image]

brandond, 2019-12-08:
The ESP talks to the Arm MCU using a variant of the flux_led protocol at 9600 bps. Unfortunately you are stuck with the present sequences and flood colors, and cannot directly access individual LEDs in the strip as you could if you were running it directly. I might add some basic support for these to ESPHome, but in the long run I'll probably end up bypassing the Arm device entirely.

yanoosh, 2019-12-15:
@brandond Do you have an idea how to bypass Arm MCU? I'm trying to connect directly to HC245, but I have a stability problem. Sometimes the whole strip works and sometimes only the first diode works. After connecting the logical analyzer it works well all the time. When I disconnect the analyzer, the problem returns after a while.

trogper, 2019-12-24:
@yanoosh it seems to me like a pull-up/down issue. Try adding a pull resistor to the lines.

VanushVaswani, 2020-02-18:
@brandond  so the ARM is the ultimate source for the WS2811 DIN pin?

brandond, 2020-02-18:
@VanushVaswani yes. Unless you want to cut and bodge-wire it you're limited to just the stuff that the ARM is preprogrammed for.

gaijinsr, 2020-03-16:
Hi, I had previously re-flashed a couple of RGB(W(W)) MagicHome controllers, so I thought the WS2811 version would be just as easy to do... After opening the controller and finding the more complex schematic, I searched for information about this controller and ended up here. Since you guys did not manage to hack it, I had to find out on my own how to do it. ;-)
Carefully following the PCB traces, I pretty soon found where to connect to the RX and TX lines, flashing the ESP was easy, but since the ESP cannot directly drive the input of a 12V WS2811 strip, I wanted to use the original 74HC245 on the board. It turned out that the circuit uses channels 1 and 3 of this chip, where channel 1 is the clock line that is not used in my version (empty top pad on the right). To feed the data signal from the ESP into the proper input of the 74HC245, I needed to connect it to the second pad on the right of the footprint of the late ARM chip.
[image]
The two lines on the bottom left are RX and TX, the upper line is connected to GPIO14 of the ESP. Works like a charm! Let me know if you need help or if you want to see my code (I use FastLED).

eogmau, 2020-03-17:
> Hi, I had previously re-flashed a couple of RGB(W(W)) MagicHome controllers, so I thought the WS2811 version would be just as easy to do... After opening the controller and finding the more complex [...]

Hi gaijinsr, thank for your explanation. Can you post a little schematic what are the three wires you have connected are doing in you mod? Are you using Espurna or other firmware and what is the pin configuration for Esp chip to the 74HC245 chip?

gaijinsr, 2020-03-17:
Hi eogmau, I think a schematic would be overkill: The two lower wires are TX (upper, IIRC) and RX. They are only needed to flash the ESP, not for running the module. (I decided to keep them in case I mess up the OTA capability.) The upper wire is connected to GPIO14 (pin 3) of the ESP, because this is where my software (I am not using Espurna) outputs the WS2811 data. (You might want to use another pin on the ESP.)
The only hardware modifications necessary are removing the ARM chip (I cut the right side of the pins with a cutter knife, twisted until the left pins broke and then removed the pins on the right with a soldering iron with some flux) and soldering a wire for the data between your chosen pin on the ESP and the second pad on the right of the former ARM chip (c.f. picture) - and of course the TX and RX lines for flashing.

eogmau, 2020-03-17:
Thanks @gaijinsr , have managed to modify two of these boards and flashed Esphome on them using the steps in your post.

brandond, 2020-03-17:
You can find a much simpler modification here:
https://github.com/Aircoookie/WLED/issues/398#issuecomment-597865330

gaijinsr, 2020-03-17:
Hi brandond, why "much simpler"? I removed one chip, they removed two resistors and part of the leads of one chip, the connection between the ESP and the resistor R8 is exactly the same in their solution and in mine. But if I had seen that post before, it would have saved me a lot of time figuring it out on my own...

gaijinsr, 2020-03-17:
Also, I think my soldering looks nicer... ;-)

brandond, 2020-03-17:
@gaijinsr because all you have to do is make one cut, vs telling people to pry off the whole ARM chip. Removing the two resistors on the UART is completely optional, it works fine without doing that but I had the iron out anyway so why not.