xoseperez/espurna #1439: wifi password with special characters..howto workaround ?

yakman000, 2018-12-19:
Dear all,
espurna works great with normal wifi password, but I'm unable to use espurna to connect with wifi which require special characters in the password (backslash ( \ )). Unfortunately I cannot change the wifi password (damn!): 

Have you got someone workaround to fix this ?

Thanks.
PS I've tested the espurna-1.13.2-itead-sonoff-basic.bin and 1.13.3

yakman000, 2018-12-19:
Thank you very much xoseperez, waiting the new-release, may you point me out how to fix (in dirty mode) ? may I hardlink the password in someone place of the code ? (certainly in this case I have to compile all the stuff).

xoseperez, 2018-12-19:
It will depend on when the problem is. There are different possibilities: storage, parsing or inside the justwifi library ESPurna uses. You can try to hardcode it in the WIFI1_PASS setting in `general.h` but I can't tell you this will make it work for you.

yakman000, 2018-12-20:
Followed your suggestion, but doesn't work.

ColinShorts, 2018-12-20:
Is it possible that the character needs to be escaped, i.e. use \\ instead
of a single \ ?

yakman000, 2018-12-20:
@ColinShorts
I made your test also, and doesn't work.

yakman000, 2019-01-11:
I made a test using the latest  espurna-1.13.4-dev.nightly20190111+gitac805701-itead-sonoff-basic.bin
still not working:
 Recap:
I have to use a password with the follow chars :"\+$%{;*!"
I'm available to test alpha/beta releases, please drop me a line, and tell me how could I debug this to you.

mcspr, 2019-01-11:
@yakman000
does get pass0 show the correct network password? or pass1, pass2 etc. if there are more networks saved
is the not-working network wpa2-enterprise/801.2x by any chance?

mcspr, 2019-01-14:
Why I was asking about wpa2-enterprise, it will not connect without specific setup code.
Cannot test this easily, but i think it will still show up when scanning.

Have you tried some basic esp8266/arduino sketches to test connection to that AP?

yakman000, 2019-01-15:
@mcspr 
I made a simple test as you suggested: 

I've used the simple sketch pubblished by randomnerdtutorials 
(https://randomnerdtutorials.com/reprogram-sonoff-smart-switch-with-web-server/).
either that is unable to connect to the wifi,that use WPA/WPA2 - Personal, therefore your idea was good, thanks, now how to move forward (?)::

The pass contains the following chars :"\+$%{;*!)" 
in this test I wrote the pass exactly as it's in the code:
const char* password = "xxx";
is it correct ? May I have to do something special to that chars ?
Are there limitations on the esp8266 libraries about this ? max lenght ? other ?

yakman000, 2019-01-17:
OK : news!
I found another AP, setup that with "WPA/WPA2 - Personal", same SSID and same passw and it works !!

Now: my diagnose is : the router has someone incompatible setup..which one ?
The router is a tp-link, googling someone has disabled wmm, other has reduced the with to 20Mhz, I tested both without success.

More: looking into the tp-link "Connection Wireless Station Numbers" I can see the mac address of the sonoff which send/receive only three packets (???), but debugging the connection using the arduino serial monitor I have the follow:::

640.Connecting to xxx
Connection status: 6
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (7): xxx
Passphrase (31): xxx
BSSID set: 0
641.Connecting to xxx
Connection status: 6
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (7): xxx
Passphrase (31): xxx
BSSID set: 0
state: 5 -> 2 (2a0)
rm 0
642.Connecting to xxx
Connection status: 6
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (7): xxx
Passphrase (31): xxx
BSSID set: 0
reconnect
state: 2 -> 0 (0)
643.Connecting to xxx
Connection status: 4
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 4
Auto connect: 1
SSID (7): xxx
Passphrase (31): xxx
BSSID set: 0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
644.Connecting to xxxx
Connection status: 6
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (7): PVnet24
Passphrase (31): xxx
BSSID set: 0
645.Connecting to xxxx
Connection status: 6
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (7): xxxx
Passphrase (31): xxxx
BSSID set: 0

---------------------
suggestions ? hints ?? the esp8266 has compatibility issues with tp-link router ??

mcspr, 2019-01-18:
@yakman000 i'd suggest to post your findings to either Espressif forums (https://bbs.espressif.com/viewforum.php?f=66) or [esp8266/Arduino](https://github.com/esp8266/Arduino) (i believe you already found the issue(s) there :) 
it looks like lower level than ESPurna, so nothing we can do about this incompatibility from our end.

You can also try Espressif's AT firmware to rule out Arduino stack as the culprit:
https://github.com/espressif/esptool
https://www.espressif.com/en/support/download/at
ESP8266 AT Bin V1.6.2 for current SDK version, install instructions inside `bin/at/README`, 1MB fw is `bin/at/512+512`
https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf, see `AT+CWJAP_CUR` command

closed 2019-01-18 by mcspr

mcspr, 2019-02-17:
@yakman000 going back a bit about not being able to change things from espurna side: can you try test sketches again and add WiFi.setPhyMode(WIFI_PHY_MODE_11G);  just before  WiFi.begin(); ?