xoseperez/espurna #1440: Depreciated warnings from home-assistant

inverse, 2018-12-19:
In the last major update of home-assistant I noticed the following warnings in the logs

```
"platform": "mqtt" is deprecated, replace with "schema":"basic"
```

From researching the issue I realise It's to do with the mqtt message with auto-discovery that will need to be adjusted.

mcspr, 2018-12-20:
Yep, it seems to be working just fine with {"schema":"basic"} addition. At first I thought this was only related to lights (searching ha github about light.mqtt_json changes), but switch component works the same.

Not really sure when deprecated becomes obsolete. No mention of it here:
https://github.com/home-assistant/home-assistant/blob/a9990c130dc613d83a63128375767458e52f4937/homeassistant/components/mqtt/discovery.py#L43
https://github.com/home-assistant/home-assistant/pull/18227 (pr that changed this)

inverse, 2018-12-20:
@mcspr thanks for adding more context :+1:

inverse, 2018-12-21:
@xoseperez that's a very valid point. I think generally people move with the releases but supporting both would be a good solution for a while and will cover people like me who keep up and when they do finally remove the feature not be broken. What do you think?

peterhoeg, 2018-12-24:
I have filed a feature request with HA for having it publish either its own version or an API version via MQTT. That would allow espurna to detect the format to use.

mcspr, 2018-12-28:
@peterhoeg Have you posted the issue to the home-assistant/home-assistant about mqtt version? I would not rely on versions for this, but it might be nice to display that ESPurna device found HA presence.

"schema" / "platform" can also be avoided altogether as it falls back to the basic mqtt method (on / off / state topics) and implicitly passes "platform": "mqtt" to the configurator.
There is no json lights support -> no need to specify it right now. And no warning.
Future version must use "schema": "json" (which is from config spec, not discovery specific) to use json payloads

peterhoeg, 2018-12-28:
@mcspr, I did on discourse as that's where they prefer feature requests: https://community.home-assistant.io/t/publish-ha-version-to-mqtt/86694

I has not received any replies though.

closed 2019-05-04 by stale[bot]