xoseperez/espurna #1443: Blitzwolf BW-SHP2 v2.3 binary

rradar, 2018-12-20:
I saw already the closed issue #1350 which is solving the software part.

What I wonder if there is a **binary image** to flash directly to the device is coming soon? The pull request for supporting the device is already merged (#1351). The Blitzwolf BW-SHP2 in revision v2.3 is right now on my table, open, soldered and ready to flash! :small_airplane: 

If the binary is not near would it make sense to flash a wrong image (for example the one for the Blitzwolf BW-SHP2) now. Than I would be able to close the device again and later use the internal updater within the web fronted to upload the proper version when released. Does it make sense? Is this possible?

Thank's for espurna to all developers and supporters and specially to @xoseperez to make it all possible! :+1:

mcspr, 2018-12-20:
🔥  https://github.com/mcspr/espurna-nightly-builder/releases/latest 🔥 
espurna-1.13.4-dev-**blitzwolf-bwshp2-v23**.bin.nightly20181220+gitbd719bd9 is the latest .bin for v2.3 (sry for naming mishap)

rradar, 2018-12-20:
hi @mcspr 
thank's a lot! Is this build compatible with the normal/stable releases? So when the next main build drops in I can easily update/change to the 'stable' build without issues (settings/factory reset is no problem)?

mcspr, 2018-12-20:
👍  Closing.

There are no restrictions in jumping up a version, to the future 1.13.4 release here. No factory reset will happen (or needed) when doing OTA upgrade, it is expected to migrate any changed settings.
Rare case when one is needed will surely be mentioned in the release notes.

closed 2018-12-20 by mcspr