xoseperez/espurna #1444: Sonoff basic new board

ralias, 2018-12-21:
There are a new board for the Sonoff Basic Model. In the board we see it labeled with the name of "SONOFF RF R2 POWER v1.0". Here you are pictures about the board. The problem is that I want a basic model to measure temperature in GPIO14 but there isn't free pin. Instead we have a GPIO2 pin prepared for use. I think that the binary espurna-1.13.3-itead-sonoff-basic-dht.bin by default doesn't work in this model. I should to modify the code to replace GPIO14 to GPIO2. Right?

[image]
[image]
[image]

nelim, 2018-12-23:
Yes, just change DHT_PIN before compile, I'm using NodeMCU with DHT on GPIO2 / D4 instead of default GPIO14.

Niek, 2019-01-10:
@ralias where did you buy this board from? I'd like to order a unit to test

ivanmarban, 2019-01-12:
@Niek I bought one of this new board from [this](https://www.ebay.es/itm/Sonoff-Basic-Smart-Home-WiFi-Wireless-Switch-Module-For-IOS-Android-APP-Ctrl/323575676326?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) seller on eBay.

Niek, 2019-01-14:
Thanks, I bought one 👍

Guillermoto, 2019-01-21:
Is there a compiled binary to use gpio2 as before as if it were gpio14?

xoseperez, 2019-02-09:
Just pushed the configuration to build the images... will be available with 1.13.4 or in the nightly builds.

closed 2019-02-10 by xoseperez

nikopolllx, 2019-03-27:
> Quizás esta imagen sea más clara:
> [image]

hola buenas amigo  dañe el boton de este modulo y ya no puedo ponerlo en modo flash como puedo puentiar el gpo0 para ponerlo a  funcionar de nuevo gracias 
[image]

Guillermoto, 2019-03-30:
No veo la imagen, pero se puede hacer un puente con un trozo de cable

Enviado el: miércoles, 27 de marzo de 2019 6:12
Para: xoseperez/espurna <espurna@noreply.github.com>
CC: Guillermo <guillermoto@ono.com>; Comment <comment@noreply.github.com>

 

 <https://user-images.githubusercontent.com/1614670/50561157-53e61d80-0d08-11e9-967b-647b816c4eeb.jpg> 

 <https://user-images.githubusercontent.com/41267456/55051848-dab74780-5024-11e9-9448-aafb3d66b73a.JPG>