xoseperez/espurna #1445: Broadlink RM Mini Espurna

seetendra, 2018-12-22:
Hello Everyone

Just wanted to see if there is a way we can flash ESPURNA on a Broadlink RIMINI.
I hate using the Broadlink application it is buggy and useless. 
Please find the pics and if there is an option by which I can flash ESPURNA 
[image]
[image]
[image]

peterhoeg, 2018-12-24:
It's a 8285 so it technically would be possible. 

I have a few of these and am using them with HA. You don't need the broadlink app for that as the python library used by HA support it.

That being said, running espurna on these babies would be awesome!

seetendra, 2018-12-24:
@peterhoeg You bet I would love to get this working with espurna. 
I have all my devices with espurna and this being the one exception.
For some reason my Rm Mini keeps going in a loop got to pairing mode then goes of the Ap Mode and then again pairing mode.
Don't know what is wrong with it any thoughts?

peterhoeg, 2018-12-26:
> Don't know what is wrong with it any thoughts?

No idea, sorry. It works fine here but I did use the broadlink app to do the initial setup as I hadn't realized at the time that the python module could handle it.

seetendra, 2018-12-26:
I have ordered a new one and requested Broadlink for a replacement.
Looks to me like an issue with the hardware then the software.

xoseperez, 2018-12-27:
I have not found any info about that module (Broadlink 3323-SBSL), @peterhoeg how do you know that's a ESP8285? I have never heard of Broadlink using Espressif chips...

peterhoeg, 2018-12-28:
@xoseperez, I remember reading *somewhere* on some forum that broadlink switched to the ESP for the mini. I have 2 of these at home as well, so I can try scratching off the stickers to see what exactly is hiding below but it does look like an ESP.

ColinShorts, 2018-12-28:
You might find that the identification mains have been helpfully scraped off, so you might need to desolder/remove the board. It does look like it could be an esp module, but I couldn't see any other modules with an exposed aerial and a square dot.
Maybe you could try backing up the flash and just blow firmware to it, but you'll have a lot of trial and error without knowing the pinout, and you still risk bricking it.

xoseperez, 2018-12-28:
Yes, a simple `esptool.py chip_id` would be enough. But I can't see the GPIO0 pad.