xoseperez/espurna #1447: Fast spin of rotary encoder reboots firmware

mcsSolutions, 2018-12-23:
**Bug description**
A slow detent-by-detent motion of rotary encoder results in expected level change behavior with limits at 0 and 255.  Fast spin of the rotary encoder results in a reboot of Espurna firmware.

Related issue is that a single detent motion results in four updates of the LIGHT_STEP.  For example if LIGHT_STEP is defined as 8, then a single detent motion from level 100 will be reflect in the Browser STATUS tab of 108, 116, 124 and then finally 132.  Expected is a single update to 108 with a single detent motion.

**Steps to reproduce**
*Steps to reproduce the behavior.*
Build magichome-20 configuration with the following added
 	    #define ENCODER_SUPPORT     1
	    #define ENCODER1_PIN1       1
	    #define ENCODER1_PIN2       3
	    #define ENCODER1_BUTTON_PIN 0   // active low by default, with software pullup
	    #define ENCODER1_CHANNEL1   0   // please note this value is 0-based (LIGHT_CH1 above)
	    #define ENCODER1_CHANNEL2   1   // please note this value is 0-based (LIGHT_CH2 above)
      // Light Dimmer
      #define LIGHT_STEP          8
Wires connected to controller RX, TX, 3.3V pads for rotary encoder CLK / DT / +.

Single detent move of encoder to produce expected result.  Spin encoder and reboot occurs.  This is reflected as the LED strip turning off, the browser page not reflecting OFF state of the LED and sometimes the WiFi SSID disconnects and then reconnects.

**Expected behavior**
No reboot with fast spin of encoder.

**Screenshots**

**Device information**
platformio run
 
Processing magichome-led-controller-20 (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)

Converting espurna.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 54 compatible libraries
Scanning dependencies...

*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*  Downloaded 2018-12-21.  espurna-dev change log update 2018-10-08
* *Flash mode* OTA
* *Device brand, model and version* https://www.amazon.com/gp/product/B07CSR6YQ9/ref=od_aui_detailpages00?ie=UTF8&psc=1

**Tools used**
* *Desktop operating system* Windows 7
* *Browser & version* Chrome
* *IDE & version* Atom/Platoform IO
* *Compiler & version (if not embedded in IDE)*

**Additional context**

mcsSolutions, 2018-12-23:
Attempting to isolate the problem by selectively commenting-out sequences has isolated the issue to be caused by the call "wsSend(_lightWebSocketOnSend)" in "void lightUpdate(bool save, bool forward, bool group_forward)" which is called from "void _encoderLoop()".

Apparently the WEB _SUPPORT has reentrancy or timing issues that surface when there are rapid callbacks from the rotary encoder motion.

For additional testing isolation a semiphore was added to _encoderLoop so that a callback is ignored if still busy servicing the prior callback.  This did not improve behavior with resets still occurring.  This seems to point to cpu use or timing issues associated with WiFi management.  

The presence or absence of MQTT_SUPPORT had no effect so this may provide some clue into likely culprit in WEB_SUPPORT.

mcsSolutions, 2018-12-23:
One additional test was to put the call "wsSend(_lightWebSocketOnSend)" inside the "if (save)" near the end of  "void lightUpdate(bool save, bool forward, bool group_forward)" to reduce the frequency of the call  to what is expected to be five seconds.  This did not improve behavior so it seems timing by itself is not the total explanation.

mcspr, 2018-12-27:
Espurna records the crash info, which can be retrieved by using `info` (reboot reason /error code) or `crash` (stacktrace) commands in terminal. You might've had easier time figuring out the reason by checking the code and then decoding the trace to pinpoint the function causing this. Unless you already did that.

Also, check if this helps: https://github.com/PaulStoffregen/Encoder/pull/15/files#diff-5c2d4a7e2bc119e539f0ad9df5ae59a6R192

xoseperez, 2018-12-27:
Probably the solution would be to slightly defer the websocket update to avoid rapid requests.

mcsSolutions, 2018-12-27:
My thought as well and that is why I tried the websocket update in the same "save" conditional block as used for reducing flash write cycles.  The issue still occurred.  I do not know how frequent the "save" condition occurs vs. encoder update, but would think it would be on the order of seconds rather than milliseconds.  I have built with conditional that bypasses websocket update if the reason for it started with encoder.  Solves the reboot problem, but some functionality lost.

mcspr, 2018-12-27:
Can you check the exception number though?

Encoder generates interrupts -> system is busy handling them. Hence the error when wifi stack tries to do the job and Encoder handler disables every interrupt (noInterrupts() etc.)
+ Note the changes from PR above that also affect this process timing in surprisingly random ways.

For the change you described - you placed it inside ticker callback function here?
https://github.com/xoseperez/espurna/blob/eb249ddaca12c297e983e913bb43b04c1d27511a/code/espurna/light.ino#L472

mcsSolutions, 2018-12-28:
Reinstalled firmware that produces undesired behavior.  Two attempts made to create failure mode.  First one able to capture crash info, but unclear if it was a reset or if it was just a loss of the socket.  The crash info was gathered after a new socket connection was made via browser.

Second attempt resulted in an actual hard crash and not able to bring it back to life with a power cycle.  I do not have ability to reflash with serial connection at this time.

Debug while knob is spun until debug no longer updates the browser window ...

---8<-------

[952137] [MAIN] ESPURNA 1.13.4-dev
[952138] [MAIN] xose.perez@gmail.com
[952139] [MAIN] http://tinkerman.cat

[952140] [MAIN] CPU chip ID: 0xC5F2DD
[952141] [MAIN] CPU frequency: 80 MHz
[952144] [MAIN] SDK version: 1.5.3(aec24ac9)
[952148] [MAIN] Core version: 2.3.0
[952151] [MAIN] Core revision: 159542381
[952155] 
[952156] [MAIN] Flash chip ID: 0x144051
[952160] [MAIN] Flash speed: 40000000 Hz
[952163] [MAIN] Flash mode: DOUT
[952166] 
[952167] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[952175] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[952181] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[952190] [MAIN] Firmware size       :   507328 bytes /  124 sectors (   1 to  124)
[952196] [MAIN] Max OTA size        :   516096 bytes /  126 sectors ( 125 to  250)
[952203] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[952210] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[952217] 
[952218] [MAIN] EEPROM sectors: 251, 250
[952221] [MAIN] EEPROM current: 250
[016954] [MQTT] Sending LightStrip/app => ESPURNA (PID 1)
[016957] [MQTT] Sending LightStrip/version => 1.13.4-dev (PID 1)
[016959] [MQTT] Sending LightStrip/board => MAGICHOME_LED_CONTROLLER_20 (PID 1)
[016962] [MQTT] Sending LightStrip/host => ESPURNA-C5F2DD (PID 1)
[017005] [MQTT] Sending LightStrip/relay/0 => 1 (PID 1)
[056362] [WEBSOCKET] Requested action: dbgcmd
[056369] [DEBUG] No crash info
[056466] +OK
[074534] [ENCODER] Delta: -1
[074538] [MQTT] Sending LightStrip/rgb => #000000 (PID 1)
[074540] [MQTT] Sending LightStrip/hsv => 0,0,0 (PID 1)
[074543] [MQTT] Sending LightStrip/mired => 326 (PID 1)
[074545] [MQTT] Sending LightStrip/channel/0 => 0 (PID 1)
[074583] [ENCODER] Delta: -2
[074586] [MQTT] Sending LightStrip/rgb => #000000 (PID 1)
[074629] [ENCODER] Delta: -1
[074632] [MQTT] Sending LightStrip/rgb => #000000 (PID 1)
[074679] [ENCODER] Delta: -2
[074727] [ENCODER] Delta: -2
[074777] [ENCODER] Delta: -2
[074825] [ENCODER] Delta: -3
[081817] [ENCODER] Delta: 1
[081820] [MQTT] Sending LightStrip/rgb => #00050D (PID 1)
[081822] [MQTT] Sending LightStrip/hsv => 217,100,0 (PID 1)
[081866] [ENCODER] Delta: 2
[081869] [MQTT] Sending LightStrip/rgb => #00050F (PID 1)
[081915] [ENCODER] Delta: 1
[081918] [MQTT] Sending LightStrip/rgb => #00060F (PID 1)
[081985] [ENCODER] Delta: 1
[081988] [MQTT] Sending LightStrip/rgb => #000610 (PID 1)
[082034] [ENCODER] Delta: 1
[082083] [ENCODER] Delta: 1
[082132] [ENCODER] Delta: 1
[082191] [ENCODER] Delta: 1
[090248] [ENCODER] Delta: 1
[090252] [MQTT] Sending LightStrip/rgb => #000815 (PID 1)
[090254] [MQTT] Sending LightStrip/hsv => 217,100,1 (PID 1)
[090257] [MQTT] Sending LightStrip/mired => 326 (PID 1)
[... 69 lines cut ...]
Edits made to encoder & light update to prohibit websocket calls due to encoder motion

void _encoderloop()
:
:
  lightUpdate(true, true, true, false); //mcs 12-23-2018 added prohibit wsSend call on encoder motion

void lightUpdate(bool save, bool forward, bool group_forward, bool wsSendEnable) {  // mcs 12-23-2018 added wsSendEnable for inhibit on encoder that causes reset
:
:
  #if WEB_SUPPORT
    
  if (wsSendEnable) {
      
    wsSend(_lightWebSocketOnSend);
    
  }
    
  #endif

void lightUpdate(bool save, bool forward) {
    
  lightUpdate(save, forward, true, true); // mcs 12-23-2018 added 4th parameter for rotary encoder inhibit of wsSend

}

mcsSolutions, 2018-12-29:
I reviewed the suggestion for "fix esp8266 crash on interrupt routines" and really do not understand what the effect of the code change is, but my understanding is the Espurna polls rather than uses interrupts for the encoder as its encoder configuration has "espurnaRegisterLoop(_encoderLoop);"

mcspr, 2018-12-29:
Encoder library uses interrupts. Loop here only reads the last state.
https://github.com/PaulStoffregen/Encoder/blob/d9c355032e4f85a1b94ae37ddf85c49b52a7f481/Encoder.h#L43 enables ENCODER_USE_INTERRUPTS by default (ESPurna does not configure this)
https://github.com/PaulStoffregen/Encoder/blob/d9c355032e4f85a1b94ae37ddf85c49b52a7f481/Encoder.h#L96-L97 implicitly attaches the interrupt handler to each pin.

BTW the PR mentioned earlier may be incomplete, as I see isr0, isr1 etc. funcs are still in flash looking at resulting .elf. Slightly changed library here, with all relevant things marked using ICACHE_RAM_ATTR.
platformio.ini / lib_deps changes to test it:
https://github.com/xoseperez/espurna/blob/2cd80ca32ead56e6405e89bbe0665c9275012507/code/platformio.ini#L83
```diff
-     Encoder
+    https://github.com/mcspr/Encoder#esp8266/isr-in-ram
```
I have not tested further than "it builds"

mcspr, 2018-12-29:
Oops. Updated for 2.3.0
`pio lib update https://github.com/mcspr/Encoder#esp8266/isr-in-ram` will update local copy
(or removing old one at .piolibdeps/Encoder* and starting build again)

mcsSolutions, 2018-12-30:
Success with build but system behavior has not changed with reboot during knob spins.  Reboot is assumed because switches are setup to set LED output to OFF on power up and the LED turns off when knob spins.  Off state is confirmed when browser connection reestablished the status shows OFF.

1st test spinning knob until debug windows stops updating
---------------------------------------------------------

[049310] [WEBSOCKET] #1 connected, ip: 192.168.0.17, url: /ws
[058627] [ENCODER] Delta: 1
[058631] [MQTT] Sending LightStrip/rgb => #040000 (PID 1)
[058633] [MQTT] Sending LightStrip/hsv => 0,100,0 (PID 1)
[058635] [MQTT] Sending LightStrip/mired => 326 (PID 1)
[058638] [MQTT] Sending LightStrip/channel/0 => 4 (PID 1)
[058677] [ENCODER] Delta: 4
[058681] [MQTT] Sending LightStrip/rgb => #140000 (PID 1)
[058726] [ENCODER] Delta: 8
[058729] [MQTT] Sending LightStrip/rgb => #340000 (PID 1)
[058776] [ENCODER] Delta: 8
[058825] [ENCODER] Delta: 9
[058875] [ENCODER] Delta: 5
[058923] [ENCODER] Delta: 5
[058972] [ENCODER] Delta: 2

New browser session established crash and info reports
------------------------------------------------------

[061390] [WEBSOCKET] #1 connected, ip: 192.168.0.17, url: /ws
[075136] [WEBSOCKET] Requested action: dbgcmd
[090548] [WEBSOCKET] Requested action: dbgcmd
[090551] [DEBUG] Latest crash was at 71566 ms after boot
[090553] [DEBUG] Reason of restart: 2
[090554] [DEBUG] Exception cause: 28
[090555] [DEBUG] epc1=0x4000bf80 epc2=0x00000000 epc3=0x4022d9e6
[090559] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[090564] [DEBUG] sp=0x3fff2550 end=0x3fff26d0
[090568] [DEBUG] >>>stack>>>
[DEBUG] 3fff2550: 3fff25d0 3fff25ac 3fff25a0 4022e385 
[DEBUG] 3fff2560: 3fff2731 00000d2c 00000d2c 4010020c 
[DEBUG] 3fff2570: 3ffe8b20 00000000 3fff2620 00000001 
[DEBUG] 3fff2580: 00000000 00000000 3fff25d0 

[140820] [MAIN] ESPURNA 1.13.4-dev
[140821] [MAIN] xose.perez@gmail.com
[140822] [MAIN] http://tinkerman.cat

[140823] [MAIN] CPU chip ID: 0xC5F2DD
[140824] [MAIN] CPU frequency: 80 MHz
[140827] [MAIN] SDK version: 1.5.3(aec24ac9)
[140831] [MAIN] Core version: 2.3.0
[140834] [MAIN] Core revision: 159542381
[140837] 
[140838] [MAIN] Flash chip ID: 0x144051
[140843] [MAIN] Flash speed: 40000000 Hz
[140845] [MAIN] Flash mode: DOUT
[140848] 
[140849] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[140857] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[140864] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[140871] [MAIN] Firmware size       :   507328 bytes /  124 sectors (   1 to  124)
[140878] [MAIN] Max OTA size        :   516096 bytes /  126 sectors ( 125 to  250)
[140887] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[140893] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[140899] 
[140901] [MAIN] EEPROM sectors: 251, 250
[140904] [MAIN] EEPROM current: 251

2nd test spinning knob back and forth until debug windows stops updating
------------------------------------------------------------------------

[178780] [ENCODER] Delta: -2
[178784] [MQTT] Sending LightStrip/rgb => #080000 (PID 1)
[178787] [MQTT] Sending LightStrip/hsv => 0,100,0 (PID 1)
[... 40 lines cut ...]
[128595] [DEBUG] >>>stack>>>
[DEBUG] 3fff2500: 00000001 0000059a 00000008 3fff2570 
[DEBUG] 3fff2510: 3fff0e04 00000033 4025f808 3fff2580 
[DEBUG] 3fff2520: 3fff2590 4020daa0 00000000 00000033 
[DEBUG] 3fff2530: 3fff0e04 3fff2570 00000000 

3rd test crash report
-----------------------

[043083] [WEBSOCKET] #1 connected, ip: 192.168.0.17, url: /ws
[060002] [MAIN] System OK
[065029] [WEBSOCKET] Requested action: dbgcmd
[075812] [NTP] Error: NTP server not reachable
[085098] [WEBSOCKET] Requested action: dbgcmd
[085106] [DEBUG] Latest crash was at 568992 ms after boot
[085108] [DEBUG] Reason of restart: 2
[085109] [DEBUG] Exception cause: 28
[085110] [DEBUG] epc1=0x4000bf80 epc2=0x00000000 epc3=0x4022da4e
[085114] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[085118] [DEBUG] sp=0x3fff25f0 end=0x3fff26d0
[085122] [DEBUG] >>>stack>>>
[DEBUG] 3fff25f0: 3fff2650 00000000 3fff2620 4010068c 
[DEBUG] 3fff2600: 00000000 000000cf 3fff2630 40207ab0 
[DEBUG] 3fff2610: 3fff2650 3fff8f40 3fff0bb0 40213542 
[DEBUG] 3fff2620: 3ffe9778 00000000 00000000

mcspr, 2019-01-03:
Thanks! At least now it is consistent in resets.

I got [PEC16-4220F-S0024](https://www.bourns.com/pdfs/PEC16.pdf) encoder to test this
Notable quirk - .read() returns data 4 times per detent (1 every loop, somehow) and therefore it triggers light reporting 4 times in quick succession. It is not as fast, avg delta i got is 17 when rotating full circle.

Instead of triggering on first read >0, I've replaced the handler to trigger on some predefined delta value accumulated from multiple reads in a row i.e. having "minimal step" before triggering any action.
Seems ok to handle full rotation with both ws and mqtt on, Windows [wscat](https://www.npmjs.com/package/wscat) as client and mosquitto running on Linux machine.
Will update with code as I figure out some weird compiler errors...

TODO related to this:
- remove settings part from lightUpdate / _lightWebSocketOnSend to slightly reduce payload size
- maybe also throttle mqtt / ws light reports using some arbitrary time value, or send some values more - often than others.
- remove light dependency of encoder to send delta via mqtt (again, reporting on interval might be useful here too)
- (?) add note about using hw debounce for A B pins for mechanical encoders

1043717432, 2019-01-09:
You've had a long day. Thank you for sharing, others behind perfect.

mcspr, 2019-06-04:
@xoseperez to summarize
- proposed changes to throttle lights update were implemented
- PaulStoffregen/Encoder#38 & PaulStoffregen/Encoder#39 were opened about the underlying cache issue. encoder function must be in ram, but is either was never there to begin with or gets dropped out of it and causes long flash load inside isr -> crash. prs are still there though.

The general idea right now is to just import encoder library here as a sensor, allowing both the use for lights and general delta values reading. Functional interrupts are kind-of a hack (and sometimes break on the same caching problem), attachInteruptArg api is not present :/

I think it is also possible to dispatching a bit different by reading gpio registers to "know" which pin triggered the change, and use a single function instead of isrN:
https://github.com/esp8266/Arduino/blob/8859b818d8557ebada8d60e76891ac7e23f5aa96/cores/esp8266/esp8266_peri.h#L51
https://github.com/esp8266/Arduino/blob/8859b818d8557ebada8d60e76891ac7e23f5aa96/cores/esp8266/core_esp8266_wiring_digital.cpp#L135
But that is likely for some other issue

PR #1769 (2019-07-07): Import Encoder library
(hopefully) resolve #1447 

- import / vendor Encoder library as local header. preserve original copyright header, add some notes (maybe this needs to be a bit more explanation there?)
- update() and isrN() now use ICACHE_RAM_ATTR, to avoid reading them from flash, causing wdt resets due to timing issues
- reduce isr function count to only 5. there is 1 isr per 2 GPIOs, and each encoder handles 2 pins
- due to the same reason, reduce isr<->encoder mapping array size
- some naming changes
- add `encMinDelta` setting
closed 2019-07-07 by mcspr