xoseperez/espurna #1449: Is the Sonoff RF bridge slow?

wsw70, 2018-12-24:
I have a Sonoff RF bridge which works fine but there is a ~1 second delay between the moment I press (in the web interface) the switch button and the action is actually performed.

On the other side is a "[smart plug](https://fr.aliexpress.com/item/5pcs-Wireless-Power-Switch-Sockets-1-Remote-Controller-Home-Mains-EU-Plug-Hot-Promotion/32623575846.html)". When switching it on/off with the associated remote, the reaction is immediate (I tried to send the signal next to the bridge to make sure obstacles do not interfere unfairly).

Could it be that the the RF bridge is simply slow to send the message (which is unfortunate but not the end of the world either), or are there tweaks I could test in order to speed this up?

**Device information**
```
[094378] [MAIN] ESPURNA 1.13.3a
[094379] [MAIN] xose.perez@gmail.com
[094380] [MAIN] http://tinkerman.cat

[094381] [MAIN] CPU chip ID: 0x8250D4
[094382] [MAIN] CPU frequency: 80 MHz
[094384] [MAIN] SDK version: 1.5.3(aec24ac9)
[094385] [MAIN] Core version: 2.3.0
[094386] [MAIN] Core revision: 159542381
[094387] 
[094389] [MAIN] Flash chip ID: 0x144051
[094392] [MAIN] Flash speed: 40000000 Hz
[094394] [MAIN] Flash mode: DOUT
[094395] 
[094396] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[094399] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[094400] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[094402] [MAIN] Firmware size       :   450704 bytes /  111 sectors (   1 to  111)
[094403] [MAIN] Max OTA size        :   569344 bytes /  139 sectors ( 112 to  250)
[094404] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[094406] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[094407] 
[094408] [MAIN] EEPROM sectors: 251, 250
[094410] [MAIN] EEPROM current: 250
[094411] 
[094412] [MAIN] EEPROM:  4096 bytes total -  2064 bytes used (50%) -  2032 bytes free (49%)
[094414] [MAIN] Heap  : 38808 bytes total - 25048 bytes used (64%) - 13760 bytes free (35%)
[094416] [MAIN] Stack :  4096 bytes total -  1488 bytes used (36%) -  2608 bytes free (63%)
[094418] 
[094419] [MAIN] Boot version: 6
[094420] [MAIN] Boot mode: 1
[094424] [MAIN] Last reset reason: Power on
[094427] [MAIN] Last reset info: flag: 0
[094429] 
[094430] [MAIN] Board: ITEAD_SONOFF_RFBRIDGE
[094432] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL WEB 
[094433] [MAIN] WebUI image: RFBRIDGE
[094435] 
[094436] [MAIN] Firmware MD5: 364173c2f02e62693792f6614b6d00b3
[094438] [MAIN] Power: 3196 mV
[094439] [MAIN] Power saving delay value: 10 ms
[094441] 
```

wsw70, 2018-12-28:
This is the effect of a switch on, then off of the lamp (via the web interface of the RF Bridge):

```
[476603] [WEBSOCKET] Requested action: relay
[476605] [RELAY] #2 scheduled ON in 0 ms
[476608] [RELAY] #2 set to ON
[476611] [RFBRIDGE] Enqueuing MESSAGE '132E009601E0E3A34A' 4 time(s)
[476613] [MQTT] Sending rfbridge-1/relay/2 => 1 (PID 1)
[477293] [RFBRIDGE] Action 0xA0
[477619] [RELAY] Setting relay mask: 36
[477796] [RFBRIDGE] Action 0xA0
[478792] [RFBRIDGE] Action 0xA0
[479071] [WEBSOCKET] Requested action: relay
[479074] [RELAY] #2 scheduled OFF in 0 ms
[479079] [RELAY] #2 set to OFF
[479082] [RFBRIDGE] Enqueuing MESSAGE '1324009601E0E3A342' 4 time(s)
[479084] [MQTT] Sending rfbridge-1/relay/2 => 0 (PID 1)
[479763] [RFBRIDGE] Action 0xA0
[480090] [RELAY] Setting relay mask: 32
[480267] [RFBRIDGE] Action 0xA0
[481263] [RFBRIDGE] Action 0xA0

```

As I mentioned:

- this delay is of about (a bit less) than a second. 
- a device directly connected to WiFi (A Sonoff Basic for instance) does not have any delays when triggered form Home Assistant: when two devices (A Sonoff Basic and a plug via Sonoff RF Bridge). are  triggered  together (the MQTT requests start at the same time), the RF based one is noticably slower
- there are no delays when triggering the plug from its remote (so it is not the plus that is slow to react)

wsw70, 2019-01-27:
@xoseperez: I moved all my switches to WiFi Sonoffs (Basic, usually) hoping that the delay between the button (wall switch) press and the on/off acion on the Sonoff WiFi switch would be shorter but it is still quite noticeable (I also has RF433 plugs before, which were relly slower to react than the WiFi switches).

In what units are the log timestamps?

```
[873733] [RFBRIDGE] Received message '2440012C03840F73EA'
[873734] [RFBRIDGE] Trying to match code 0F73EA
[873742] [MQTT] Sending rfbridge-1/rfin => 2440012C03840F73EA (PID 1)
```

There is `9` "something" difference between the 'received RF signal' and 'sent MQTT message' - I feel like this being about a second so that could be 900 ms.

xoseperez, 2019-01-27:
Timestamps are in milliseconds, so this is actually 9ms... If you feel it's something around a second maybe we should start looking at the broker...

wsw70, 2019-01-27:
I do not think this i sthe MQTT broker. When activating a switch from Home Assistant (which sends a message to MQTT to which the switch is subscribed to) it is instantaneous.

I will work on a complete log, from each of the elements, to exactly pinpoint the place the delay comes from.

wsw70, 2019-02-02:
Sorry for the noise @xoseperez . I did all kinds of tests and I think this is the nature of these automations to be slightly slower than a "real" switch. 

I could not exactly pinpoint the component (wall switch vs. propagation vs. RF Bridge vs. MQTT vs. Home Assistant vs. Sonoff Basic) but I think that each contributes a bit to the slight delay.

closed 2019-02-02 by wsw70

padcom, 2020-03-29:
I have a number of those sonoff devices around the house. The MQTT connectivity is ultra fast with the RF434-MQTT bridge but some of the devices react so fricken slow it is almost unbearable. Even the GPIO0 switch has an over 500ms delay (I'd say it feels like 1s more than 500ms). I am on the latest 1.14.1 of Espurna. Is this kind of delay normal?