xoseperez/espurna #1453: AllJoyn Support

debarko, 2018-12-27:
**AllJoyn Support on ESP8266.**
*ESPurna supports multiple service discovery protocols like mDNS, LLMNR, NETBIOS and SSDP. Moving forward towards a AllJoyn based service discovery helps create a much needed definition for such services and a more stable discovery protocol. AllJoyn with it's intention for IOT service discovery and Qualcomm backing would really be stable and future proof.*

**Describe the solution you'd like**
*Create an AllJoyn client so that it can respond to all the queries for such service discoveries.*

**Additional context**
*[AllJoyn](https://en.wikipedia.org/wiki/AllJoyn) *

debarko, 2018-12-27:
Seems like alljoyn developers dropped the ball and nothing is moving on their repos. Windows 10 IOT has dropped their support. In fact currently their website's SSL certificate is also expiring. Now I am not sure as to whether this support is needed or not.

But this protocol is really good.

xoseperez, 2018-12-27:
There has been no real activity for more than a year. It doesn't either look "simple" to implement. I don't think this is worth it.

closed 2018-12-27 by xoseperez

mcspr, 2018-12-27:
Little sidenote:
AllJoyn is merged with IoTivity project, but HW page suggest the most support is given for much faster platforms than ESP8266 https://wiki.iotivity.org/hardware
WIP support (arduino core based as it seems?) for esp was suggested, but no movement there: https://gerrit.iotivity.org/gerrit/#/c/13315/
One can go around implementing protocol without their's SDK, but not that is not as "simple" as with it.