xoseperez/espurna #1457: MQTT not appearing

patrol420, 2018-12-28:
* *Arduino Core version*
* *ESPurna version* 1.13.3
* *Flash mode* dout
* *Device brand, model and version* generic lightbulb tw3eys wifi chip with esp8266

**Tools used**
win10 and ide 1.8.7

ive flashed 1 of these already and everything was seemless. this one i cannot seem to get the MQTT option to appear in web menu. ive probly built it 20 times trying different things in hardware.h for mqtt support. but nothing works. output on seriel on startup says 
[000674] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000687] [THINGSPEAK] Async ENABLED, SSL DISABLED
seems to be on but not appearing. any ideas

xoseperez, 2018-12-28:
Maybe a problem in the web image? Are there any javascript errors in the browser?

patrol420, 2018-12-28:
doesnt seem to be, if i flash the latest bin for lc02 i get every option in the menu... im i correct in thinking that if i was to compile the latest source with the hardware selected i should have mqtt default. i remove // before mqtt support in arduino.h and still same thing.

xoseperez, 2018-12-28:
MQTT is enabled by default in all builds. So you should have this. You can also check if you are receiving a `mqttVisible` key via web sockets on the browser (using the network inspector).

kakaki, 2018-12-28:
I have similar problem with Schedule, ntp and Sensors menus - they are not showing - got 2 identical BW-SHP2 - one is on 1.13.1 - working fine with all options visible in menu, other was upgraded to 1.13.3.
In web sockets there is no info to set ntpVisible, shdVisible and sensorsVisible.

Same problem is on other devices updated to 1.13.3.

[image]
[image]

patrol420, 2018-12-28:
> I have similar problem with Schedule, ntp and Sensors menus - they are not showing - got 2 identical BW-SHP2 - one is on 1.13.1 - working fine with all options visible in menu, other was upgraded to [...]

going off what you said, i just flashed 1.13.1 and it works perfect first go. dam i spent 4 hours on it yesterday just flashing it, shes held up to atleast 100 flashs lol

mcspr, 2018-12-30:
What lwip version are you using (v1, v2, low memory, high bandwidth)? Assuming it is self-built from "win10 and ide 1.8.7"
Can you check with [this version](https://github.com/me-no-dev/ESPAsyncWebServer/archive/4c621f340c731d621e21fef4d32928f4bec61b85.zip) of ESPAsyncWebServer lib?
Or [this](https://github.com/me-no-dev/ESPAsyncWebServer/archive/313f3372c61f6f4bff9c157f02a07ba912d3d4d9.zip), basing on platformio.ini pinned version. Only needed if using webui on iOS

kakaki, 2018-12-30:
Thanks, it helps, was using version 1.2.0 from git - it's never then version in archive, add's message eques.
https://github.com/me-no-dev/ESPAsyncWebServer/commit/bed4146ec647fd1cdfd1d2485724369c92cab8e6

reaper7, 2018-12-31:
I have the same issue with unvisible menus for mqtt, shedule, ntp...
arduino 1.8.7, win10, lwip high bandwidth

espurna from master branch (1.13.3) works OK
espurna from dev branch (1.13.4-dev) works but **without** mqtt,schedule,ntp in webmenu

I don't change any external libs, only espurna versions

xoseperez, 2019-01-04:
Can you build the web interface images using `gulp` command? Check this wiki page: https://github.com/xoseperez/espurna/wiki/WebInterface

xoseperez, 2019-01-04:
I have rebuilt and commit the webUI images, maybe you can also do a `git pull`, rebuild and flash.

mcspr, 2019-01-04:
> arduino 1.8.7, **win10**, lwip high bandwidth, **ESPAsyncWebServer current git version**

my bet it is related to #1300. current git of espasyncwebserver limits the buffer of sent ws messages by 8. that buffer is kept until the client receives them and windows is inherently slow at that.
+ as confirmed by @kakaki, rolling back a version helps

there is a workaround by doing request-response for module configuration, that way client will send N config requests and receive N config responses back, bu i have not tried it with lwip1 builds.

reaper7, 2019-01-08:
@mcspr - I have not tried the older version of espasyncwebserver yet
because I use the latest version in several other projects.
 
I build 1.13.3(working) and 1.13.4-dev(not working) with the same espasyncwebserver(current master version),
so ... something has changed inside espurna when it comes to using a web server.

mcspr, 2019-01-12:
@patrol420 Thank you for providing config!

As to difference between 1.13.3 and 1.13.4-dev:
If you are using Chrome, you can check web inspector to see that it sends exactly 9 ws messages on 1.13.3. Which fits in above limit, because first is acked immediately while 8 are still queued.
1.13.4-dev revamps relay config to send more ws messages, which results in 12 messages that no longer fit the queue fully.
And this could be safely mitigated by not using latest commit to the master of ESPAsyncWebServer. Queue limit added seems arbitrary, as it based on amount of messages and not their size, and can safely be avoided here.

I think i am going to do the proposed workaround to avoid possible issues like this (+ there would be nicer api to fetch module config after the initial one)

reaper7, 2019-01-12:
@mcspr - thanks for explaining :)
I'm new espurna user, I did not know this repo deeply yet. (compared to tasmota, it's damn fast, I'm going to stay with this project)

You understand that I would like to have and use one version of the external library like ESPAsyncWebServer for all "my" projects.
so...every possible improvement will be appreciated.

daxkamala, 2019-01-24:
Hooray! And thanks for the pointers.
I just changed line 30 in AsyncWebSocket.h (#define WS_MAX_QUEUED_MESSAGES) from 8 to 16 and all the menus returned.
This file will be under your libraries, ESPAsyncWebServer-Master in my case.
Everything else seems to work fine now.

reaper7, 2019-01-25:
@daxkamala - I checked and it works :)

mcspr, 2019-02-21:
@reaper7 @daxkamala can you verify changes in #1387 ? 
without modifying WS_MAX_QUEUED_MESSAGES value of ESPAsyncWebServer

pre-built webui headers (replace `code\espurna\static\*.h`):
[webui.zip](https://github.com/xoseperez/espurna/files/2887540/webui.zip)

reaper7, 2019-02-21:
@mcspr - not yet, I will try during the weekend. (if arduino will reconcile with python ;) )

mcspr, 2019-02-21:
@reaper7 two options:
1. https://help.github.com/articles/checking-out-pull-requests-locally/
2. slightly more involved. in the espurna git repo directory:
```
$ git remote add mcspr https://github.com/mcspr/espurna
$ git fetch mcspr compact-ws
$ git checkout -b compact-ws mcspr/compact-ws
```
or, if you do not want to modify your files, instead of checkout:
```
$ git worktree add -b compact-ws compact-ws mcspr/compact-ws
# compact-ws/ directory will now contain files from PR
```

reaper7, 2019-02-21:
@mcspr - looks wery well!
for me almost everything works as it should...
with one small exception -> SWITCHES (SONOFF T1 (3) )
Switches bookmark shows 11 switches, without numbers:
[image]

I always use the same custom config:
```
/*
my devices:
0 sonoff rf bridge  192.168.0.150
1 sonoff t1 3ch     192.168.0.151
2 sonoff t1 3ch     192.168.0.152
3 sonoff t1 3ch     192.168.0.153
4 sonoff t1 3ch     192.168.0.154
5 sonoff t1 3ch     192.168.0.155
6 sonoff rf         192.168.0.156
7 sonoff rf         192.168.0.157
8 sonoff s20        192.168.0.158
9 sonoff s20        192.168.0.159
*/

#define SEL_DEVICE 1

#if SEL_DEVICE == 0
  #define ITEAD_SONOFF_RFBRIDGE
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonoffbr150"
  #define RELAY_BOOT_MODE             0                                         // 0 - off, 1 - on, 2 - same
  #define LED1_MODE                   LED_MODE_RELAY
  #define WIFI1_IP                    "192.168.0.150"
#elif SEL_DEVICE == 1
  #define ITEAD_SONOFF_T1_3CH
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonofft13151"
  #define RELAY_BOOT_MODE             2                                         // 0 - off, 1 - on, 2 - same
  #define LED1_MODE                   LED_MODE_FINDME
  #define WIFI1_IP                    "192.168.0.151"
#elif SEL_DEVICE == 6
  #define ITEAD_SONOFF_RF
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonoffrf156"
  #define RELAY_BOOT_MODE             2                                         // 0 - off, 1 - on, 2 - same
  #define LED1_MODE                   LED_MODE_FINDME
  #define WIFI1_IP                    "192.168.0.156"
#elif SEL_DEVICE == 7
  #define ITEAD_SONOFF_RF
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonoffrf157"
  #define RELAY_BOOT_MODE             2                                         // 0 - off, 1 - on, 2 - same
  #define LED1_MODE                   LED_MODE_FINDME
  #define WIFI1_IP                    "192.168.0.157"
#elif SEL_DEVICE == 8
  #define ITEAD_S20
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonoffs20158"
  #define RELAY_BOOT_MODE             2                                         // 0 - off, 1 - on, 2 - same
//  #define LED1_MODE                   LED_MODE_FINDME
  #define LED1_MODE                   LED_MODE_MQTT
  #define WIFI1_IP                    "192.168.0.158"
#elif SEL_DEVICE == 9
  #define ITEAD_S20
  #define MY_ROOM                     "salon"
  #define MY_HOSTNAME                 "sonoffs20159"
  #define RELAY_BOOT_MODE             2                                         // 0 - off, 1 - on, 2 - same
//  #define LED1_MODE                   LED_MODE_FINDME
  #define LED1_MODE                   LED_MODE_MQTT
  #define WIFI1_IP                    "192.168.0.159"
#elif SEL_DEVICE == 99
//  #define ARILUX_AL_LC11
[... 67 lines cut ...]
  #define NTP_SUPPORT                 1                                         // Build with NTP support by default (6.78Kb)
  #define NTP_SERVER                  "myhome.lan"

  #undef MQTT_SUPPORT
  #define MQTT_SUPPORT                1                                         // MQTT support (22.38Kb async, 12.48Kb sync)
  #undef MQTT_ENABLED
  #define MQTT_ENABLED                1                                         // Do not enable MQTT connection by default
  #define MQTT_AUTOCONNECT            1                                         // If enabled and MDNS_SERVER_SUPPORT=1 will perform an autodiscover and
  #define MQTT_SERVER                 "rpi"                                     // Default MQTT broker address
  #define MQTT_USER                   "mqttuser"                              // Default MQTT broker usename
  #define MQTT_PASS                   "mqttpass"                              // Default MQTT broker password
  #define MQTT_RETAIN                 true                                      // MQTT retain flag
  #define MQTT_QOS                    0                                         // MQTT QoS value for all messages
  #define MQTT_USE_JSON               0                                         // Group messages in a JSON body

  #define HOMEASSISTANT_SUPPORT       1                                         // Build with home assistant support (if MQTT, 1.64Kb)
  #define HOMEASSISTANT_ENABLED       1                                         // Integration not enabled by default
  #define HOMEASSISTANT_PREFIX        MY_HOME                                   // Default MQTT prefix

  #define DOMOTICZ_SUPPORT            0                                         // Build with domoticz (if MQTT) support (1.72Kb)
  #define THINGSPEAK_SUPPORT          0                                         // Enable Thingspeak support by default (2.56Kb)
  #define ALEXA_SUPPORT               0                                         // Enable Alexa support by default (10.84Kb)

#endif
```

xoseperez, 2019-02-21:
Just merged #1387 which solves this issue within ESPurna. Anyway it will be interesting to have some different approach to this on AsyncWebSocket class...

closed 2019-02-21 by xoseperez

mcspr, 2019-02-21:
@reaper7 i think you were missing correct webui. courtesy of Xose, `dev` now has prebuilt ones, so just `git pull` will do. i do not see the bug you describing rn with a test device

xoseperez, 2019-02-22:
Yes, I think web UI images should be created for every new feature committed. It's not obvious you have to build them too otherwise.

reaper7, 2019-02-22:
You're probably right, I'll check it in the evening again :)

mcspr, 2019-02-22:
> Yes, I think web UI images should be created for every new feature committed. It's not obvious you have to build them too otherwise.

True. My main problems are diffs and general repo size effects.

While I was writing this, turns out this can be easily solved via .gitattributes to treat them as binary blobs:
```
*.gz.h    -diff -merge
*.gz.h    linguist-generated=true
*.gz.h     -diff -merge
*.gz.h     linguist-generated=true
```

*I would also add a evil check to the CI, comparing generated files to the ones in the commit and fail if they are not the same*

ref:
https://help.github.com/en/articles/customizing-how-changed-files-appear-on-github
https://git-scm.com/docs/gitattributes#_marking_files_as_binary

reaper7, 2019-02-22:
it was a problem with missing correct webui :) mea culpa !
tnx!

schweini, 2019-03-09:
I am still missing the MQTT web interface.

i just did a `git pull` and the web interface says i'm running ESPURNA 1.13.4-dev .
I'm using the library `async-mqtt-client-master` from the .zip linked to on the wiki page.

the relevant parts from my custom.h are:

```
#define MQTT_ENABLED                1               // Do not enable MQTT connection by default
#define MQTT_USE_JSON               1               // Group messages in a JSON body
#define MQTT_SERVER                 "XXXXXXX"
#define MQTT_USER                   "XXXXXXX"              // Default MQTT broker usename
#define MQTT_PASS                   "XXXXXXXX"              // Default MQTT broker password
#define MQTT_USE_ASYNC              1           // Use AysncMQTTClient (1) or PubSubClient (0)

```
I have also tried using MQTT_USE_ASYNC 0, but no change.