xoseperez/espurna #1458: Tuya devices

germaguire, 2018-12-28:
Has anyone looked at tuya device support for espurna? All of the wifi devices have an esp8266ex board TYWE1S so I presume they are candidates for reflashing.
https://docs.tuya.com/en/hardware/WiFi-module/wifi-e1s-module.html

Im still looking for a more definitive list of products but it looks interesting.
https://en.tuya.com

Ger.

xoseperez, 2018-12-28:
These are no different from any other ESP8266 module. TheTeckin SP22 v1.4+ uses a Tuya TYWE2S and is supported in the current `dev` branch.

[image]

Wikibear, 2019-01-15:
Question will be answered here:
https://github.com/xoseperez/espurna/issues/1431

closed 2019-01-20 by xoseperez

germaguire, 2019-01-28:
Some interesting developments re not having to solder Tuya devices in order to flash espurna. Have to wait until this evening to test but I have some new light bulbs, extension socket and wall plugs ready to try out.
https://github.com/ct-Open-Source/tuya-convert

https://www.youtube.com/watch?v=O5GYh470m5k&feature=youtu.be

forast3r, 2019-05-15:
First of all, my apologies for posting on a closed issue.
Can't thank @xoseperez enough for the great work with espurna firmware.

@End0rphi is the push button on the kloe4 power strip working?
Flashed Espurna 1.13.5 on a kloe4 with tuya-convert but the button only turns off the power strip after keeping it pushed for 10 or more seconds. However haven't been able to switch on any of the relays with the button. Does the boot mode of the individual switches influence in this behaviour? Shouldn't the button toggle (or at least switch all on or off)?

End0rphi, 2019-05-15:
I can confirm that in the default configuration for this device, the button does not switch the relays. This is expected behavior according to [the device entry in the wiki](https://github.com/xoseperez/espurna/wiki/Hardware-Zhilde-ZLD-44EU-W).

I'm quite sure the button behavior can be changed with the -DBUTTONx_MODE C preprocessor flags documented [here](https://github.com/xoseperez/espurna/wiki/Buttons-and-switches), but I have not tried it

forast3r, 2019-05-15:
Thank you for the confirmation and the links to the wiki!