xoseperez/espurna #1462: Test sonoff rf bridge

krovakav, 2019-01-02:
Hello there, having lot of time during Christmas I've came with idea of power consumption measuring (with some 433mhz module). I have sonoff bridge, I've modified it to read codes directly with esp8266, but now I need to test it. 
I have some old wireless bell button (silvercrest? I am not sure if it is working) and few xiaomi temperature sensors. So there should be some communication in the air. But... Espurna behaves like there is no communication at all. How can I get raw data in console (or anywhere) or is there any common way to test rf communication?
Thank you

xoseperez, 2019-01-04:
That is somewhat complicated. You should first know what frequency they are using and then the encoding and protocol. For Xiaomi devices I think they use ZigBee, so no luck there. About the wireless bell button, I can't tell.

krovakav, 2019-01-04:
I think this was wrong question from me.
Firstly I want to ask if there is some possibility to display communication through 433 mhz (as raw bytes)
Second question should be what (possibly simplest/cheapest) device could  be used to test the device/espurna?

And as I was thinking about my solution, I would like to make two pulse counters to monitor electricity and gas consumption by counting pulses (electric meter has blinking diode, gas probably magnetic). 
So the implementation would be some simple Arduino which will count pulses and every N pulses it will send actual value by rf (I could use esp and wifi, problem is power consumption). 
And here comes the issue - does espurna support some protocol to do that? Or maybe - how to connect custom device to espurna?

xoseperez, 2019-01-05:
A custom device will probably require custom integration. If your device uses a simple RF link you might start building something on top of the RFBridge module.

Another option is to use the RFM69 module which I use to monitor battery-powered devices at home using Hope RFM69 radios. You can check my [RFM69 Gateway](http://tinkerman.cat/rfm69-wifi-gateway/) post about it. In the post I used a custom firmware for the gateway but now I use ESPurna with the RFM69 module enabled.

closed 2019-01-10 by krovakav