xoseperez/espurna #1463: Sonoff Pow R2 Questions

rradar, 2019-01-02:
Hello everybody,

I have three Questions about the [Sonoff POW R2](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-POW-R2)

**Question 1**
*Does espurna save the consumption measurements 'offline' (inside the flash memory from the esp?). If so - is there a way to see the historical data or does it just store the total consumption?
For sure to use the gatherd data correctly a mqtt broker, home assistant, influxdb and grafana would be a elaborated scenario but I'm wondering if espurna also could give some stats for a standalone/off grid installation*

**Question 2**
*Can it be that the [Sonoff POW R2](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-POW-R2) has a total of 4MB flash? (I read so somewhere but can't find the source anymore) If so - does the the provided binary for this device does make use of it?*

**Question 3**
*This device has [unused GPIO (io4 and io5)](https://github.com/xoseperez/espurna/issues/1126). I know it's possible to use it for [sensors](https://github.com/xoseperez/espurna/wiki/Sensors) supported by espurna (for example DHT22). Do I need to compile a custom firmware for it or can I use the one provided in the release section for this propose?*

I think I found the answer for question 3 already! Good to have a wiki :smile: 
> This page describes various sensors supported by ESPurna.
> 
> To enable particular sensor in firmware, you should build the firmware with appropriate build flags (-D) set to 1. The table below lists all supported build flags. Some sensors have additional options, so please check additional information below.
https://github.com/xoseperez/espurna/wiki/Sensors

Bon any nou! :fireworks:

rradar, 2019-01-02:
An additional question:

I found in the [Sonoff-Tasmota Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow-and-Pow-R2) the following information:

> 
> DO NOT CONNECT ANY SENSOR TO THESE DEVICES. Only use them as designed.
> 
> The GPIOs on the Pow are connected to AC power, therefore it is not possible to add sensors to the Pow.
> 

Is it true? It's not possible to connect sensors to the sonof pow r2 io's? :confused: :question:

ColinShorts, 2019-01-02:
It would not be //safe// to connect external sensors. There is a good chance that it would not meet local safety standards, it would pose a serious risk of electrocution, and/or fire. It may pose problems for your home insurance.

Possible, yes. Sensible, absolutely not.

xoseperez, 2019-01-04:
The ground in the POW is connected to neutral mains, so it's not safe to do that.

ESPurna does not store data offline, it only saves the latest value for the energy so it can resume after a reboot. SPIFFS is not loaded by default so there is no logging feature, but it is, of course, something that can be added.

AFAIK POW R2 has 1MiB flash chips, but it is true they sometimes report 4MiB. I have not tested it.

rradar, 2019-01-09:
Thank's @xoseperez all questions answered!

Going to close this issue

closed 2019-01-09 by rradar

HobbytronicsPK, 2019-01-17:
I have tested numerously Pow R2 switches and they have 4MB flash chips

giuaig, 2019-01-19:
Shouldn't be better to add a disclaimer on the POW-R2 wiki page about the risk of connecting unused gpio 4/5 to sensors, due to the AC power?
I would edit that page myself but i don't know if i'm allowed to...