xoseperez/espurna #1470: Problem after OTA Upgrade

Alferez, 2019-01-04:
I have updated a Sonoff Touch from 1.13.3 to 1.13.4-dev (previously I had already had other versions).
The problem is that after updating it does not connect to the network.
I have made a hard reset and it remains the same, it is put in AP mode because the network does not see it. If you sign in and do a network search there it is but it does not connect even if you reboot, it always enters AP mode.
I update them by generating a new bin with itead-sonoff-touch-ota and uploading it by web.
The same thing happened to me with a T1-2ch and with other Touch.
Logically I understand that it is something that I am doing wrong, because it seems that this only happens to me.

Regards.

xoseperez, 2019-01-05:
Do you mean that when you flashed via OTA it stopped connecting to the network but if you do it uploading the image via the web interface it worked?

Alferez, 2019-01-05:
No, device work without problem for 3 month. I try Alexa detect and for it I need upgrade.
I generate ota firmware with Plataformio and I upload it from web interface.
When upgrade finished, device restart and in this moment start in AP mode.
I access and network are correct configured, but every time device start in AP mode. 
If reset at Factory default, same issue, start in AP mode when network are correct configured.

xoseperez, 2019-01-18:
Can you copy here the info of the device as stated [here](https://github.com/xoseperez/espurna/blob/dev/.github/ISSUE_TEMPLATE/bug_report.md)?

closed 2019-03-26 by stale[bot]