xoseperez/espurna #1480: build fails with web_support=0 for secure mqtt

ankycooper, 2019-01-08:
I tried both master & dev, but the build fails with web_support=0 works fine with web support.
I'm trying to get secure mqtt to work which requires web to be disabled.
I'm using PLATFORMIO

Here is the log

Compiling .pioenvs\nodemcu-lolin\lib297\ESPAsyncTCP\AsyncPrinter.cpp.o
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino: In function 'bool _alexaBodyCallback(void*, uint8_t*, size_t, size_t, size_t)':
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:33: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:52: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:66: error: 'HTTP_GET' was not declared in this scope
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:83: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino: In function 'bool _alexaRequestCallback(void*)':
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:87:27: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:87:62: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:33: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:52: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:66: error: 'HTTP_GET' was not declared in this scope
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:83: error: 'void*' is not a pointer-to-object type
// Global switch
^
Compiling .pioenvs\nodemcu-lolin\lib297\ESPAsyncTCP\ESPAsyncTCP.cpp.o

ankycooper, 2019-01-09:
Tried disabling Alexa support with the following :
`
[env:nodemcu-lolin-ota-mqtt-ssl]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_4m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_4m1m} -DNODEMCU_LOLIN -DDEBUG_FAUXMO=Serial -DNOWSAUTH -DUSE_CUSTOM_H=1 -DWEB_SUPPORT=0 -DALEXA_SUPPORT=0 -DASYNC_TCP_SSL_ENABLED=1 -DMQTT_USE_ASYNC=1
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = ${common.upload_flags}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}
`
custom.h has admin password, wifi and mqtt settings

I get the following warning and build fails:

`Linking .pioenvs\nodemcu-lolin-ota-mqtt-ssl\firmware.elf
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o):(.bss.default_private_key+0x0): multiple definition of `default_private_key'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):(.bss.default_private_key+0x0): first defined here
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o):(.bss.default_private_key_len+0x0): multiple definition of `default_private_key_len'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):(.bss.default_private_key_len+0x0): first defined here
c:/users/v2ank/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: Warning: size of symbol `default_private_key_len' changed from 2 in .pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o) to 4 in .pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o)
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o):(.bss.default_certificate+0x0): multiple definition of `default_certificate'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):(.bss.default_certificate+0x0): first defined here
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o):(.bss.default_certificate_len+0x0): multiple definition of `default_certificate_len'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):(.bss.default_certificate_len+0x0): first defined here
c:/users/v2ank/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: Warning: size of symbol `default_certificate_len' changed from 2 in .pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o) to 4 in .pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o)
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o): In function `SSLContext::getIOContext(int)':
C:\Users\v2ank\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp:183: multiple definition of `ax_port_read'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):c:\Users\v2ank\espurna\code/.piolibdeps\ESPAsyncTCP\src/tcp_axtls.c:545: first defined here
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o): In function `SSLContext::getIOContext(int)':
C:\Users\v2ank\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp:183: multiple definition of `ax_port_write'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):c:\Users\v2ank\espurna\code/.piolibdeps\ESPAsyncTCP\src/tcp_axtls.c:489: first defined here
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o): In function `ax_get_file':
C:\Users\v2ank\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp:568: multiple definition of `ax_get_file'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):c:\Users\v2ank\espurna\code/.piolibdeps\ESPAsyncTCP\src/tcp_axtls.c:470: first defined here
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib6e2\libESP8266WiFi.a(WiFiClientSecure.cpp.o): In function `ax_wdt_feed':
C:\Users\v2ank\.platformio\packages\framework-arduinoespressif8266@1.20300.1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp:612: multiple definition of `ax_wdt_feed'
.pioenvs\nodemcu-lolin-ota-mqtt-ssl\lib297\libESPAsyncTCP.a(tcp_axtls.c.o):c:\Users\v2ank\espurna\code/.piolibdeps\ESPAsyncTCP\src/tcp_axtls.c:584: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\nodemcu-lolin-ota-mqtt-ssl\firmware.elf] Error 1
================================================================= [ERROR] Took 26.25 seconds =================================================================`

Please help

ankycooper, 2019-01-14:
Thanks for the reply, there is some progress.

Build still fails with 242, but works with ALEXA_support disabled 

`Compiling .pioenvs\nodemcu-lolin-mqtt-ssl\lib098\ESP8266WiFi\WiFiServer.cpp.o
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino: In function 'bool _alexaBodyCallback(void*, uint8_t*, size_t, size_t, size_t)':
CC:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:79:33: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:79:52: error: 'void*' is not a pointer-to-object type
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:79:52: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:79:66: error: 'HTTP_GET' was not declared in this scope
alexa.createServer(false);
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:79:83: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino: In function 'bool _alexaRequestCallback(void*)':
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:87:27: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:87:62: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:89:33: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:89:52: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:89:66: error: 'HTTP_GET' was not declared in this scope
// Global switch
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:89:83: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino: In function 'bool _alexaBodyCallback(void*, uint8_t*, size_t, size_t, size_t)':
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:81:1: warning: control reaches end of non-void function [-Wreturn-type]

^
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino: In function 'bool _alexaRequestCallback(void*)':
C:/Users/xxx/Desktop/espurna/code/espurna/alexa.ino:91:1: warning: control reaches end of non-void function [-Wreturn-type]

^
*** [.pioenvs\nodemcu-lolin-mqtt-ssl\src\espurna.ino.cpp.o] Error 1`

Any fix?

mcspr, 2019-01-15:
I think so. Can you try this patch? #1491

PR #1491 (2019-01-15): Use built-in fauxmoESP HTTP handler when WEB_SUPPORT=0
fix #1480
closed 2019-01-15 by xoseperez