xoseperez/espurna #1481: Sonoff Bridge switch status

speedyrazor, 2019-01-09:
Hi, I have just flashed my new Sonoff Bridge with version 1.13.3, and installed an RF light switch in my kitchen (https://www.amazon.co.uk/gp/product/B01G42AAMA/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1). The RF code is the same for on and off, and all is working fine. BUT, if I manually switch the light on by the light switch, ESPURNA still reports the status as off, if I then switch it on in ESPURNA, the light turns off, so it is now back to front.
Is there anyway round this please?

speedyrazor, 2019-01-10:
perhaps building an Arduino mini with RF receiver and transmitter?
I wrote another post here to explain more:
https://www.reddit.com/r/arduino/comments/aei899/arduino_mini_rf_relay_uk_light_control/

Regards.

DocFingo, 2019-01-17:
@wildwiz would you mind testing this case with your modified rfbridge hardware please? If it's a EFM8BB1 thing then this could be the solution.

wildwiz, 2019-02-06:
Hi Doc, the whole point to understand what is going on is to have the real remote talking to a modified (RFDIRECT) bridge. Once you have that you can either dump the RF timings, with a specific debug firmware, or just try and see if it works. 

Unfortunately, I don't have that kind of remote myself.
If you are just interested in RF RX and you don't need RF TX, then you can modify the bridge in such a way that it can, depending on the firmware installed, work both with EFM8BB1 and direct. 

You will be able to transmit only with EFM8BB1 firmware, but you can complete the mod later if that's the case.

What you have to do is OMIT the "Transmitter signal rewiring" mod and just do "Receiver signal rewiring" and "USB data lines deactivation (ONLY FOR R2 BOARDS)". If you go this way, the device is RX compatible with both flavors of Espurna firmware (and with the diagnostic firmware) and you can swap back and forth.

You can find the diagnostic firmware here: https://github.com/xoseperez/espurna/pull/864
This firmware is able to dump raw received timings, which may help further investigating.

xoseperez, 2019-02-07:
I think @wildwiz is right. If it uses the same code for ON and OFF, and there is a human action somewhere (you said you manually switched the light), then ESPurna will be out of sync. Same would happen if it misses any RF message generated by a third party, like the original remote.

There's no way around this, unless you are monitoring somehow the real status of the light and reporting the RF Bridge about it in a way it does not trigger a message again...

DocFingo, 2019-02-09:
@xoseperez , indeed monitoring the real status is mandatory for an RF toggle switch. I have some non-toggling devices which I'm using wherever appropriate, works like a charm.

A couple more observations:  With two identically programmed RF handsets, I can switch the same set of lights and outlets, but the RFbridge sees _different_ yet very similar codes for each.  The lamps don't care, the RFbridge does. Weird. 

Second observation, ESPurna sends each RF code multiple times, which is not always helpful for toggle switches when the code pause is long enough. Interstingly, relay channel 7 sends enough ON/TOGGLE RF codes to blink the lamp 4 times (pause ~350 ms). But relay channel 0, using the _same_ RF codes, only toggles the lamp once - and between ch1 and ch6 it somehow gets increasingly worse.  Weird ^2.

xoseperez, 2019-02-09:
ESPurna sends the same code only if the codes for ON and OFF are different. Codes are enqueued for sending, there should be no difference in timing... this is something we should check. Can you copy here the log you get when sending the codes?

DocFingo, 2019-02-11:
Relay channel 0, sending RF code 26AC015E03C0144154 (identical for ON and OFF)
```
[239565] [WEBSOCKET] Requested action: relay
[239568] [RELAY] #1 scheduled ON in 0 ms
[239571] [RELAY] #1 set to ON
[239574] [RFBRIDGE] Enqueuing MESSAGE '26AC015E03C0144154' 1 time(s)
[240448] [RFBRIDGE] Action 0xA0
[240576] [RELAY] Setting relay mask: 135
```
(EOM)

Relay channel 7, sending RF code 26AC015E03C0144154 (identical for ON and OFF)
```
[122234] [WEBSOCKET] Requested action: relay
[122237] [RELAY] #7 scheduled ON in 0 ms
[122240] [RELAY] #7 set to ON
[122243] [RFBRIDGE] Enqueuing MESSAGE '26AC015E03C0144154' 4 time(s)
[123108] [RFBRIDGE] Action 0xA0
[123251] [RELAY] Setting relay mask: 135
[123609] [RFBRIDGE] Action 0xA0
[124117] [RFBRIDGE] Action 0xA0
[124617] [RFBRIDGE] Action 0xA0
```
(EOM)

**Solution**: Since you wrote ...

> ESPurna sends the same code only if the codes for ON and OFF are different.

... I suspected some invisible character stored with the RF string that made them compare differently, although that difference did not appear in the send log.  I "forgot" both codes and programmed them again (ON=OFF=26AC015E03C0144154), and now I'm getting the expected behaviour: 
```
[877115] [WEBSOCKET] Requested action: relay
[877118] [RELAY] #7 scheduled ON in 0 ms
[877122] [RELAY] #7 set to ON
[877125] [RFBRIDGE] Enqueuing MESSAGE '26AC015E03C0144154' 1 time(s)
[877999] [RFBRIDGE] Action 0xA0
[878128] [RELAY] Setting relay mask: 135
```

Presumably the same happened for other channels as well, with contradicting symptoms. I was using copy/paste over whatever was stored there before. 

Thanks!

xoseperez, 2019-02-11:
Good. Closing this then, since for the original message there is no way to know the real status of the switch from ESPurna.

closed 2019-02-11 by xoseperez