xoseperez/espurna #1483: Counter sensor freeze

pkedvessy, 2019-01-11:
Hi,

Using counter sensor gets freeze my SONOFF Dual R2 device after a while when sending singlan frequently to the board. Seems like the variable to measure ticks is overflowing or something like that.

I'm on the master branch, environment is itead-sonoff-dual-r2-ota and additional build flags are the followings:
`-DEVENTS_SUPPORT=1 -DEVENTS_PIN=3 -DEVENTS_PIN_MODE=INPUT -DEVENTS_INTERRUPT_MODE=RISING`

Without having any sensor connected to the board I don't experience any issue.

pkedvessy, 2019-01-12:
Debug is turned off by default for Sonoff Dual R2 by default. Isn't is?

mcspr, 2019-01-14:
Nope. See
https://github.com/xoseperez/espurna/blob/9dbd14e38892ee2d0c901406e65050a462b5f855/code/espurna/config/hardware.h#L487-L512

pkedvessy, 2019-01-14:
Ok, I had a try with the suggested build flag, but the result is the same.

mcspr, 2019-01-15:
...which is? does it reboot or stops responding? "count" reported will reset each report, this is expected

you can also try setting `EVENTS_TRIGGER=0` to avoid reporting individual events

pkedvessy, 2019-01-15:
Which is `-DDEBUG_SERIAL_SUPPORT=0`
It stops responding and WIFI status led stops blinkink.

xoseperez, 2019-01-20:
Do you have the chance to monitor the debug log to see what happens?

closed 2019-03-28 by stale[bot]