xoseperez/espurna #1486: Ability to have multiple different type sensors

kris969, 2019-01-12:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

flashy02, 2019-01-15:
Hi,
normaly it is possible to use more than one sensor on the same note. For my usage I've tried a SHT31 and a BME280 at one I²C. On an other note I tried an analog pressure Sensor + a SHT31 and a Dallas OneWire. Both nodes work fine.

Regards

xoseperez, 2019-01-20:
I can't see why it shouldn't work unless you are using the same GPIOs for the DHT22 and I2C.

kris969, 2019-01-26:
I was not using the same GPIOs, I took care of this.
It seems that it was an electrical issue. I was using +3V form LoLin board, I changed power connections, and everything is now working fine.
I have to read in detail Lolin specifications to improve my understanding, maybe I have done something wrong using power connections.
Thanks for replies.

xoseperez, 2019-02-07:
Cool. Closing this

closed 2019-02-07 by xoseperez