xoseperez/espurna #1488: Magichome 20 Binary

chr2002qw, 2019-01-13:
Hello,

i flashed a Wemos D1 Mini with the Magichome 2.0 prebuilt binary, brutforced the R G B Gpio Pins and got my LED Strip working with OpenHab2.

Now i want to use the Switch feature (GPIO0) i connected a switch to this GPIO Pin, setup MQTT but nothing happens. Theres no output about a pressed button in the Debug screen in the Espurna Web interface.
If i connect the switch to GPIO2, the onbord Led of my Wemos lits up upon pressing the button until i release it again.

Why theres no info about the button event in the debug screen ? What iam doing wrong ?
I tested it with high and low signals.

ColinShorts, 2019-01-14:
Could you try using another gpio pin instead of 0/2?

chr2002qw, 2019-01-14:
I just tested every GPIO with high 3,3V and Low GND without sucess. Theres no info about a button event in the Debug screen.

closed 2019-03-28 by stale[bot]