xoseperez/espurna #1493: Adjusting brightness: Relative setting

jbaumann, 2019-01-17:
While it is possible to set the brightness to an absolute value (using MQTT or the REST API), a setting relative to the current brightness value is not possible. I would like to e.g., make a lamp brighter by a value of 25 (without having to know the current brightness).

Misiu, 2019-01-18:
To support this MQTT message should look like this:

{root topic}/brightness/set | +25 | add 25 to current value (if more that 255 then 255 is set)
-- | -- | --

This is possible, but if You have MQTT and openHAB for example, then You can calculate new value based on current value.
With REST this requires 2 requests (first to get value, second to set new one).

What do You think about supporting payload with plus or minus sign? This way Your requirement can be solved.

jbaumann, 2019-01-19:
For me this solution would work perfectly well. I'm using MQTT and NodeRed and keeping state in NodeRed when it is in the Espurna firmware already is kind of error prone e.g., if the NodeRed-instance is restarted the state is lost (I assume the same happens with OpenHAB).
From a design viewpoint of such a system this system shows a much better behaviour (more resilient and degrading more graceful) if these increments/decrements could be handled relative to the values stored locally in the devices... Well, my opinion at least.

mcspr, 2019-11-09:
Introduced by https://github.com/xoseperez/espurna/pull/1607 and extended via https://github.com/xoseperez/espurna/pull/1938
Every input now accepts + and -

Closing.

closed 2019-11-09 by mcspr