xoseperez/espurna #1497: Power usage statistics? QUESTION

evlo, 2019-01-19:
Hi, 

I have lot of sonoffs and I flashed them to tasmota, however tasmota so I don't need app to control them.
I found out that tasmota does not have power usage statistics so I don't know how much power is used each month.
Sonoff default firmware has it, however I now don't have any option to put back original firmware.

My question is if espurna does have power stastistics functionality

I know I can get third party solution running home assistant or grafana or whatever on server, but it would be nice and more robust and stable if I could get switches to collect power usage statistics on their own.

xoseperez, 2019-01-20:
I'm not 100% sure but I'd say the original firmware does statistics in the cloud since storing them locally would have a big impact on flash life expectancy. ESPurna does not store locally any data except for the current total energy consumption, and that's only optional (disabled by default).

xoseperez, 2019-02-02:
4MB is of course enough to store data for weeks or months, more if you have a way to aggregate and remove old data. Problem is that each time you save, even 1 single byte, the whole sector gets written and that wears out a lot the memory. That's why I thing they should be using a different approach...

closed 2019-04-10 by stale[bot]